Stephan3/dwc2-for-klipper #2: [FR] Touch friendly interface

Hywelmartin, 2019-02-25:
Touch friendly interface would be nice.
I'm using a 5" phone as my main screen at the printer

hevilp, 2019-02-25:
Hi,

please ask your question on https://github.com/chrishamm/DuetWebControl/tree/next
where DWC is developed. The fork connect DWC to klipper and does not change anything to the Webif itself.

Hywelmartin, 2019-02-25:
sorry for my bad google fu...

closed 2019-02-25 by Hywelmartin

Stephan3, 2019-02-26:
it works great on my ipad mini btw

Hywelmartin, 2019-02-26:
I found a closed FR at duetwebcontrol and they referred to dwc2 as a version that was more touch friendly..
I will test when your PR get merged to Klipper Main branch..

If you want to post screen shots from your phone (I haven't found any from small screens ) that would be great...

Stephan3, 2019-02-26:
Seriously? You ask me to make screenshots as you´re to lazy to make this 5 minutes installation?

Hywelmartin, 2019-02-26:
I take your warnings seriously.. even that I think they aren't  that seriously meant..

> !!! CAUTION !!!
> Its an early stage, horrible things can happen to you, your printer, your moms cellar or your cat. For first trys its a good idea to limit printer speed in config to a low value like 25mm/s or 50mm/s. If a ${disastercondition of doom} is occouring, you can hit the emergency button. The emergency button shuts down immediately no gcode queue or similar in between, it just overrides everything.

I asked and you refused ... so I will, as I wrote, test it when your PR get merged....

over and out

/Martin

Stephan3, 2019-02-26:
https://github.com/Stephan3/dwc2-for-klipper/commit/cda1b6d758420a47561fd781ce2f4e4e4437b995

jakep82, 2019-02-26:
I installed last night.  Took maybe 10 minutes.  I literally just copy/pasted the install instructions into a terminal.  I installed on an existing Octopi/Klipper install, so I also disabled Octoprint.  Almost everything works flawlessly.