Stephan3/dwc2-for-klipper #3: I want this in my life

JamieAKing, 2019-02-25:
This looks amazing, I'm new to 3d printing and coming from the cnc router control interfaces my first impression was how the hell do people deal with octoprint. I find it very  counter intuitive and the need to keep moving between tabs is annoying to say the least.
my biggest annoyance is the lack of a machine position read out, so no way to know where the tool head is at a glance 
the front end of dwc looks fantastic, so much info and control in one place.

Please do not take this as a criticism but simply a sign of my ignorance , I have read the install procedure 3 times and have absolutely no idea as to what I need to do. 

I have a pi3b+ running a stock install of octopi and klipper installed as per Kevins instructions for klipper.
how do I make the change? assume you are talking to an carpenter with access to a keyboard

many thanks in advance

hevilp, 2019-02-25:
Hi,

you are wrong here, this software connects DWC to klipper.
The readme is full everything you need to install. If you don't understand, you need to read more.

jakep82, 2019-02-25:
The [instructions](https://github.com/Stephan3/dwc2-for-klipper#on-octopi--ubuntu--debian-untested-feedback-wanted) seem pretty clear.  SSH into your Pi, and copy/paste 1 line at a time.  Then add [these lines](https://github.com/Stephan3/dwc2-for-klipper#klipper-config-example) to your printer.cfg file.  You will also need to either install the patch from the pull request, or manually edit the gcode.py file as indicated [here.](https://github.com/KevinOConnor/klipper/pull/1290/commits/f13010708fbc9db9e0e586d592dfc1d7ed6aac81)

If you aren't comfortable with this, then you will probably need to wait until the development is done, and DWC support is merged into Klipper.  This is still WIP.

hevilp, 2019-02-25:
There is a high chance, that the fork never will be merged into Kevins sources...

BlackStump, 2019-02-25:
I can not see why Kevin would not add the few lines to gcode.py as really that is all that is needed.
DWC is a separate entity no different to Octoprint so nothing to merge into Klipper.
The way Stephan3 has installed the web_dwc2.py is smart as it is easy to update so it matters not if Kevin chooses to not change the gcode.py. not difficult to add the lines yourself.

Stephan3, 2019-02-26:
then go there and tell him to merge it 💃 

https://github.com/KevinOConnor/klipper/pull/1290

closed 2019-02-26 by Stephan3

JamieAKing, 2019-03-16:
Followed all of the steps but octoprint the is still loading In place of dwc

BlackStump, 2019-03-16:
Octoprint is your ip:5000  eg 192.168.1.42:5000
if you followed the steps with dwc
DWC will be your ip:4750  eg 192.168.1.42:4750

if you want to change the port then it is in this section in your config
[web_dwc2]
# optional - defaulting to Klipper
printer_name: Reiner Calmund
# optional - defaulting to 127.0.0.1
listen_adress: 0.0.0.0
# needed - use above 1024 as nonroot
listen_port: 4750    <=======
#	optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web

JamieAKing, 2019-03-16:
Perfect thank you