Stephan3/dwc2-for-klipper #4: Webcam stream

vekenti, 2019-02-28:
I got the webcam stream to work pretty easily.

If one used octopi and installed klipper over it, then everything is already set up and ready to go, but it is also pretty easy to set up if a stock raspberry pi image without octopi is used. Octopi uses MJPG-Streamer to stream the webcam. By installing ocotpi it already sets up the streamer to start on boot so the user doesn't have to themselves. As explained by this guide, https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337 (farther down there is a webcam section), making MJPG-Streamer start on boot is really quite easy.

Once the streamer is running, all one has to do is under User Interface>General Settings> Webcam URL, put in the URL of the stream ( It will be the same one you access the DWC with, but change the port) and then add /?action=stream to the end. Octopi by default has the stream at port 8080, and if you set MJPG streamer manually then use whatever port you set it to.

[image]

Example. I access DWC with 192.168.0.4:4750. My camera URL is 192.168.0.4:8080/?action=stream.

Stephan3, 2019-02-28:
Yes this will work. BUT i am allready on an Solution to run this tornado thread on another core. On a raspi2 i had issues with the current solution. The code will be pulled this weekend, thanks.

Stephan3, 2019-03-01:
support is there now and works for me without crashes -> closed

closed 2019-03-01 by Stephan3