Stephan3/dwc2-for-klipper #6: Gcode hanging issue

shaiss, 2019-04-09:
I'm trying out dwc w/klipper and so far a few issues.  
1. Some gcodes when entered in dwc's gcode console will execute but dwc spins forever thinking the gcode never ran.  Ie "m117 test" will display test, but a subsequent "m117 hello" will do nothing.  I'm watching the klippy.log but there's no particular error.

Stephan3, 2019-04-10:
The ack system is just very different in rrf and klipper. Thats why i do a pseudo ack on a set of commands. i added m117 to them. Just checkout latest master, test and report back.

shaiss, 2019-04-11:
I'll pull and test it out.  I noticed it with other commands like testz
when calibrating.  And a few others.  I'll report back.

Stephan3, 2019-04-12:
yes, please add a list

shaiss, 2019-04-15:
@Stephan3 I made the change manually per commit d14302117acab17a6f26f7dbca5accd7d7bbbc48 adding M117 and it works solving this issue.  I'll be pulling a list from the klipper gcodes doc @ https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md.  particularly TESTZ

Stephan3, 2019-04-15:
Those, klipper has no feedback needs to be there. I dont use TESTZ, so i did not realized that.

shaiss, 2019-04-16:
commit 07b48f0b2331a2bac7d7214fa8791557f3d00d48 fixes this issue, closing ticket as original m117 hanging issue is resolved and more gcodes will be added in the future.

closed 2019-04-16 by shaiss