Stephan3/dwc2-for-klipper #7: Extrusion control disabled

shaiss, 2019-04-09:
Extrusion control is disabled.   Even when the extruder is at 220 (above the min extrusion temp) and g1 e100 f300 works, these buttons in dwc are disabled.

Stephan3, 2019-04-10:
try latest master. It should work now. it lso adds multitool support.

shaiss, 2019-04-11:
Awesome and ty!

shaiss, 2019-04-11:
@Stephan3 I just did a pull and am now getting an error re current_tool.  I presume an update is required to the printer.cfg?

`Warning:  Klipper state: Shutdown
4/11/2019, 3:49:25 PM: STATUS: Warning:  Internal error during ready callback: web_dwc2 instance has no attribute 'current_tool'
Warning:  Once the underlying issue is corrected, use the
Warning:  "FIRMWARE_RESTART" command to reset the firmware, reload the
Warning:  config, and restart the host software.
Warning:  Printer is shutdown
Warning:  Klipper state: Not ready
Error:  Internal error during ready callback: web_dwc2 instance has no attribute 'current_tool'`

mtsgsd, 2019-04-12:
I found this as well. I just updated Klipper  which appears to have change it's enumeration of the tools. Maybe this is why ?
`Uncaught exception GET /rr_status?type=1 (192.168.1.111)
HTTPServerRequest(protocol='http', host='192.168.1.114:4750', method='GET', uri='/rr_status?type=1', version='HTTP/1.1', remote_ip='192.168.1.111')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 247, in get
    self.web_dwc2.rr_status_1()
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 663, in rr_status_1
    "currentTool": self.current_tool,   #       must be at least 1 ! learned the hardway....
AttributeError: web_dwc2 instance has no attribute 'current_tool'`

Stephan3, 2019-04-12:
Have you guys tries to read that pile of words on your own? Please post a log next time in readable format. Thanks

try latest master now

shaiss, 2019-04-12:
@Stephan3 I pulled the master but now seems the klipper console goes through and endless restart cycle.  I noticed "AttributeError: web_dwc2 instance has no attribute 'sdcard'" in the logs.  pictures attached for better readability.
[image]
[image]

Stephan3, 2019-04-12:
I need to do it on my machine. It is in parts currently

shaiss, 2019-04-12:
@Stephan3 np, let us know how we can help testing.

mtsgsd, 2019-04-12:
Just an FYI, I have the rolling messages, but no errors in the log.

Stephan3, 2019-04-15:
I rolled back to earlier version for now. Theese reconnects happen when dwc webapp errors. you can see the log of dwc when you press strg+shift+i in your browser. theres a console opening then shows whats happening.

shaiss, 2019-04-15:
@Stephan3 looks like issue #8 is the proper place for the reconnect problem @mtsgsd and I are reporting.

If we go back to the root of this issue, extrusion control being disabled, is that fixed in the latest commit and we should verify then close this ticket, or is that still an issue?

not trying to pressure, just trying to help make sure we have these issues cleaned up.

Stephan3, 2019-04-15:
The confusion basicaly was initiated by pushing untested code. my printer(voron 2) works with the current version on github.

Stephan3, 2019-05-01:
I am late to the party but try the current master pls. @shaiss

shaiss, 2019-05-01:
Will do. I'll git pull and let you know!

Stephan3, 2019-05-27:
@shaiss  any news? should work? Can i close this?

shaiss, 2019-05-27:
Crud I forgot. Let me check tomorrow and get back to you.

closed 2019-06-07 by Stephan3