Stephan3/dwc2-for-klipper #8: Constant re-connects using DWC

tripstar76, 2019-04-13:
As title.
I've attatched 2 log files. 
Latest is the latest DWC2 for klipper, and 3e98053 is the last commit that worked(Fixes the "lets set heater to -273.15")
The errors in the latest log file show up as soon as the DWC webpage is opened, and then the constant stream of reconnects which blocks all user control to the webpage.

[klippy.3e98053.log](https://github.com/Stephan3/dwc2-for-klipper/files/3076402/klippy.3e98053.log)
[klippy.latest.log](https://github.com/Stephan3/dwc2-for-klipper/files/3076403/klippy.latest.log)

Stephan3, 2019-04-15:
I rolled back to earlier version, can you test again pls?

tripstar76, 2019-04-15:
Hi Stephan, I'll test it when I've finished work. I maybe should have put in my original post that I reverted to the '-273' commit to get DWC2 back up and running.

shaiss, 2019-04-15:
@tripstar76 is your issue similar to mine @ https://github.com/Stephan3/dwc2-for-klipper/issues/7#issuecomment-482550319?

mtsgsd, 2019-04-15:
just pulled the latest version and I get this error:
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 130, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 235, in check_unused_options
    section,))
Error: Section 'web_dwc2' is not a valid config section

tripstar76, 2019-04-15:
@Stephan3 
`Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 126, in _read_config
    self.try_load_module(config, section_config.get_name())
  File "/home/pi/klipper/klippy/klippy.py", line 109, in try_load_module
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1232
    <<<<<<< HEAD
     ^
SyntaxError: invalid syntax
^C
`

tripstar76, 2019-04-15:
@Stephan3 
[https://pastebin.com/7TiAAqcq](url)

Stephan3, 2019-04-15:
@mtsgsd we had that earlier on discord. the static link is missing. 
@tripstar76 its fixed on latest master. Was a entry from the rollback. It works now on my machine again.

tripstar76, 2019-04-16:
Hi @Stephan3,

It's back printing again, thank you very much.
But, I think the temp filter might not be working,  thats if it's to stop the stream of toast popups while the printer is heating up? For me, this is just cosmetic, I'm happy that you've got the connection problem working! :)

tripstar76, 2019-04-16:
I think this is corrected

closed 2019-04-16 by tripstar76