Stephan3/dwc2-for-klipper #17: 3 extruders, one hotend

dystechnic, 2019-05-27:
Hi,

First of all, Thanks for the work! I love this, this is so much more informational and faster than Octoprint.

I have a printer with 3 extruders and a 3-in 1-out switching hotend. So I only use one heater shared over the 3 extruders. So I don't need 3 hotends showing up in the web-interface. 
Is it possible not only to check in printer.cfg if there are multiple extruder definitions but also check if they share the same heater? In printer.cfg this is configured as `shared_heater: extruder0` in the [extruder1] and [extruder2] section.
I that is the case you could only show extruder0 and omit the other extruders.

Thanx.

Stephan3, 2019-05-27:
Hi.

Basicaly everything is possible. I thing the 2 extruders are listed with invalid temperatures or do they all have the same temperature?

- Stephan

dystechnic, 2019-05-27:
Hi,

All extruders show the same temperature.

Ed.

Stephan3, 2019-09-30:
as i am not going to work on this, this will close.

closed 2019-09-30 by Stephan3