Stephan3/dwc2-for-klipper #19: linux host shutdown command

BlackStump, 2019-06-17:
Is it possible add a shutdown and reboot command so as to shutdown the host from within DWC
ie sudo shutdown -h now
ie sudo shutdown -r now

also babysteps is not working  MOVE=1 needs to be added as per change log 20190520
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md

Thank you

dystechnic, 2019-06-25:
I second that request. That would be a nice enhancement.

zmrde, 2019-06-29:
I also would like to see run native linux commands.

Stephan3, 2019-06-30:
I Agree with the MOVE=1 thing you mentioned. But why do we need native Linux commands? Next thing i am affraid of is, the day we merge, this might become a problem.

BlackStump, 2019-06-30:
If it is any concern with a merge then I agree leave it out for now!
The reason I asked is at the moment I use DWC then when I want to shutdown the printer I get out of DWC and open up Putty to log in to the Orangepi enter username and password issue the "sudo shutdown -h" enter password again and wait for the shutdown and power off. 
I have been doing this for sometime so I can continue but it would be nice to be able to shut it down from DWC .
I would say that is the only thing I miss from Octoprint.

Thanks again for your work
Cheers

manu7irl, 2019-07-14:
maybe parsing the command from a macro could be possible

Stephan3, 2019-08-13:
I dont think that i will make that happen. You should ask Kevin if klipper should support hostcommands by gcode.

closed 2019-08-13 by Stephan3