Stephan3/dwc2-for-klipper #20: Height Map co-ordinates do not match bed width

Oddy9923857, 2019-06-30:
I have a 310x310 print bed and when I probe the bed for a mesh it is displayed in the height map page; however if I click on a point it is not the correct X & Y co-ordinate. Otherwise the height map is correct, just it seems to get the X & Y co-ordinates off (like it is scaling to a smaller bed size (100x100).

Stephan3, 2019-08-13:
Tested today on a 300x300 Voron 2 and a 350x350 voron 2 and works for me. So not sure what it is.
Once you load your map, klipper dwc2 creates a heightmap.csv If you can grab that file, we can take a look at it.

closed 2019-09-30 by Stephan3

lenne0815, 2019-12-15:
having the same problem, the csv seems to be correct as far as i can tell though ?
[heightmap.zip](https://github.com/Stephan3/dwc2-for-klipper/files/3964869/heightmap.zip)
Max z Probed point is 74.5 in the gui, even though the csv says 224.98