Stephan3/dwc2-for-klipper #21: gcode uploaded but nor printing

zmrde, 2019-06-30:
Hello. Does anyone experieced this issue? Thanks

Stephan3, 2019-06-30:
Remove white space from file names .

zmrde, 2019-06-30:
There are no white spaces in the file name

Stephan3, 2019-06-30:
Then take a look into log file and find the spot where you try to start print

Scottbg1, 2019-07-12:
Stephan, I am having the same problem as zmrde is having. I looked in what I think you were referring to as the log, if DWC logs in the same place as Klipper, being /tmp/klippy.log, to see if I could determine the trouble, and admittedly I do not know exactly what I am looking at.

I will paste what I think you might would need to diagnose, but first a bit of a more detailed description of what is going on. 

When uploading a gcode file, via the "UPLOAD & START" button, the file loads into the gcode directory but the job does not start automatically. Nothing happens as if the "START" part of "UPLOAD & START" is not functioning correctly. You can however, go into the gcode directory and manually start the job from there and everything is fine. I thought maybe it was conflicting with some gcode that Simplify3D was inserting but it does the same with Cura as well.

This is a newly installed version of Klipper and DWC onto a raspberry pi 3b+ runinng Raspbian Stretch.
DWC v2.0.0-RC5
No Octoprint installed.

Here is what I copied from the klippy.log that I think is the beginning steps when the file is loaded:

If this isn't what you need to find the trouble, please let me know I will post it.

```
M32 /gcodes/Filament Guide v4.gcode
Stats 46.7: gcodein=0 mcu: mcu_awake=0.024 mcu_task_avg=0.000154 mcu_task_stddev=0.000165 bytes_write=3104 bytes_read=7273 bytes_retransmit=36 bytes_invalid=0 send_seq=191 receive_seq=191 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.0$
Uncaught exception GET /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22 (192.168.1.65)
HTTPServerRequest(protocol='http', host='192.168.1.9:4750', method='GET', uri='/rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22', version='HTTP/1.1', remote_ip='192.168.1.65')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1590, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 228, in get
    self.repl_ = self.web_dwc2.rr_gcode( self )
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 538, in rr_gcode
    handover = func_(params)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1079, in cmd_M32
    raise 'gcodefile' + fullpath + ' not found'
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1079, in cmd_M32
    raise 'gcodefile' + fullpath + ' not found'
  File "/usr/lib/python2.7/bdb.py", line 49, in trace_dispatch
    return self.dispatch_line(frame)
  File "/usr/lib/python2.7/bdb.py", line 68, in dispatch_line
    if self.quitting: raise BdbQuit
BdbQuit
DWC2:02019.95 - 192.168.1.65 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22
M32 /gcodes/Filament Guide v4.gcode
Uncaught exception GET /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22 (192.168.1.65)
HTTPServerRequest(protocol='http', host='192.168.1.9:4750', method='GET', uri='/rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22', version='HTTP/1.1', remote_ip='192.168.1.65')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1590, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 228, in get
    self.repl_ = self.web_dwc2.rr_gcode( self )
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 538, in rr_gcode
    handover = func_(params)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1079, in cmd_M32
    raise 'gcodefile' + fullpath + ' not found'
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1079, in cmd_M32
    raise 'gcodefile' + fullpath + ' not found'
  File "/usr/lib/python2.7/bdb.py", line 49, in trace_dispatch
    return self.dispatch_line(frame)
  File "/usr/lib/python2.7/bdb.py", line 68, in dispatch_line
    if self.quitting: raise BdbQuit
BdbQuit
DWC2:02019.97 - 192.168.1.65 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22
M32 /gcodes/Filament Guide v4.gcode
Uncaught exception GET /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22 (192.168.1.65)
HTTPServerRequest(protocol='http', host='192.168.1.9:4750', method='GET', uri='/rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FFilament%20Guide%20v4.gcode%22', version='HTTP/1.1', remote_ip='192.168.1.65')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1590, in _execute
```

Scottbg1, 2019-07-12:
I figured it out. It didn't click with me about white spaces in the file name but that was it. I renamed the file with no spaces and it worked properly.

closed 2019-08-13 by Stephan3