Stephan3/dwc2-for-klipper #22: Section dwc2_web is not a valid config section

Scottbg1, 2019-07-16:
Stephan, I started an issue thread on the Klipper hub about my problem as I thought my issue might have been with Klipper alone. But now I am not sure.

I am trying to install Klipper with DWC onto an RPi running Arch.
I have it running successfully on Raspbian after following your instructions but after repeating the same steps more times than I can count on Arch I am running into errors. I think Klipper is running properly but only assume that it is if I completely remove the [dwc2_web] section from my .cfg.
It's only then that Klipper begins communicating with the printer.
If I re-enable that section, I get `Section dwc2_web is not a valid config section` and communication is lost. 
And it goes without saying I have yet to be able to connect to the DWC UI.
Thanks for your attention to this.

```
Starting Klippy...
Args: ['/opt/klipper/klippy/klippy.py', '/etc/klipper/klipper.cfg', '-l', '/var/log/klipper/klippy.log', '-I', '/run/klipper/sock']
Git version: 'v0.7.0-691-g0af5b9ff-archlinux'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Mar 15 2019, 13:03:02) \n[GCC 8.2.1 20181127]'
Start printer at Sun Jul 14 20:58:53 2019 (1563155933.5 970.9)
===== Config file =====
[virtual_sdcard]
path = /home/scott/sdcard

[dwc2_web]
printer_name = Anet A8
listen_adress = 0.0.0.0
listen_port = 4750
web_path = dwc2/web

[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -20
position_max = 205
position_min = -23
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -20
position_min = -23
position_max = 205
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = .0025
endstop_pin = ^!PC4
position_endstop = 0
position_max = 240
homing_speed = 10

[extruder]
step_pin = PB1
dir_pin = PB0
enable_pin = !PD6
step_distance = .0105
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA7
control = pid
pid_kp = 2.151492
pid_ki = 0.633897
pid_kd = 230.042965
min_temp = 0
max_temp = 250
pressure_advance = 0.4

[heater_bed]
heater_pin = PD4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA6
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 1000
max_z_velocity = 20
max_z_accel = 100

[display]
lcd_type = hd44780
rs_pin = PA3
e_pin = PA2
d4_pin = PD2
d5_pin = PD3
d6_pin = PC0
d7_pin = PC1
=======================
Extruder max_extrude_ratio=0.266081
Config error
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/opt/klipper/klippy/klippy.py", line 131, in _read_config
    pconfig.check_unused_options(config)
  File "/opt/klipper/klippy/configfile.py", line 235, in check_unused_options
    section,))
Error: Section 'dwc2_web' is not a valid config section
```

Stephan3, 2019-07-16:
You have a path Problem. Where ever you try to link the py file it’s wrong

Scottbg1, 2019-07-16:
This is the path, per your install instructions:
`ln -s ~/dwc2-for-klipper/web_dwc2.py ~/klipper/klippy/extras/web_dwc2.py`

Stephan3, 2019-07-16:
You are brave enough for arch then please check where your Klipper Installation is, where you cloned dwc and what this ln does...

Scottbg1, 2019-07-16:
From what I understand about Arch, Klipper is located in the /opt directory.
I cloned DWC to the user directory.
And I know that ln creates a dynamic link.

So with that in mind, I removed web_dwc2.py from ~/klipper/klippy/extras/.
Then-
`ln -s ~/dwc2-for-klipper/web_dwc2.py /opt/klipper/klippy/extras/web_dwc2.py`

I confirmed that web_dwc2.py is now located in /opt/klipper/klippy/extras/web_dwc2.py.
I restarted Klipper but I still get the same error.

I got to think about the config sections for dwc in the Klipper.cfg file and how the paths were laid out.
```
[virtual_sdcard]
path: /home/scott/sdcard

[web_dwc2]
# optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web
```
The virtual sdcard path compared to the web_dwc2 web path should be relative since the 'dwc2' and 'web' directories are located in '/home/scott/sdcard', not to mention the telltale description in the web_dwc2 config section. 
So I moved the web_dwc2.py from the dwc2-for-klipper folder to the sdcard folder and ran:
 `ln -s ~/sdcard/web_dwc2.py /opt/klipper/klippy/extras/web_dwc2.py`
Still the same error message.

I'm at a loss.

Scottbg1, 2019-07-19:
I finally found some help on another site and a guy named Wilriker assisted me with getting it going. 
It seemed to be a permissions problem with the sym-link between the 2 `web_dwc2.py` files that we couldn't get sorted out. 

He suggested putting the `dwc2-for-klipper` directory in the `/opt` directory along with the `klipper` directory and it works perfectly now. He even went as far as writing an install script for the DWC bits that I plan to use to create a `PKGBUILD` similar to the install script for Raspbian to help other people out that might be having issues getting it to work.

So this indeed was NOT a klipper issue. My apologies for posting here.

closed 2019-07-19 by Scottbg1