Stephan3/dwc2-for-klipper #26: Show saved Gcodes in Klipper display

mbgroot, 2019-08-13:
The Klipper searches for files in the CD card (~/sdcard) and does not show the files in the subdirectories. 
DWC2 stores the files in /gcodes (~/sdcard/gcodes) and then does not show them in the klipper on the display.
Can I change this?

mbgroot, 2019-08-13:
I tried to change the _/DuetWebControl/src/utils/path.js_ file to _gcodes: '0:/gcodes',_ to _gcodes: '0:/',_ but then the folder navigation logic in the file manager is broken.

Stephan3, 2019-08-13:
This is like klipper does that. For me in 2019 there is no reason to drop subfoldersupport. If you wand you can hardlink that as a workarround.

~/dwcstuff (give that as sdpath in dwc config)
~/sdcard (hardlink to ~/dwcstuff/gcodes)

mbgroot, 2019-08-13:
I was thinking about the hard link...
just a little bit confused about where to find the dwc config? 
And what variable is responsible for sdpath?

Stephan3, 2019-08-13:
I am talking about the config block in your klipper config.

mbgroot, 2019-08-13:
hmm... but:
`Error: Option 'sdpath' is not valid in section 'web_dwc2'`

closed 2019-09-30 by Stephan3