Stephan3/dwc2-for-klipper #27: port to replicape

killisch, 2019-10-17:
Is anyone interested in working on porting DWC2 for Replicape to replace octoprint?

i don't have the skillset myself but would happily test or help out where possible.

Stephan3, 2019-10-17:
Sorry but i dont have the knowledge of replicape to do so.

BlackStump, 2019-10-17:
@killisch  Replicape is just hardware and is already supported by Klipper and works with @Stephan3 wonderful translator.
Your question would be better asking if porting to redeem which is the firmware created by the Replicape guys, and to be blunt poorly implemented and supported imho you would be better just going with klipper.
I suspect the answer will be no different in anycase.

killisch, 2019-10-18:
Understood. Feel free to close this issue.

closed 2019-10-21 by killisch