Stephan3/dwc2-for-klipper #30: ack needers and midprint_allow command - needs more commands in

manu7irl, 2019-11-01:
Hi Stephan,
thanks for this amazing DWC translator,
I got 2 issues that are actually related,
adding manually every single command, needed is hard to do,
could it be done simply by parsing all the commands klipper offers?
And arranging the list in:
1- What Klipper command or Gcode command does not need ack, then load them in ack_needers array
2- What Klipper command or Gcode command nees to be used while printing, then load them in the midprint_allow array
If not
I was thinking about some sort of parsing from the printer.cfg file under:

```
[web_dwc2]
# optional - defaulting to Klipper
printer_name: YOUR_PRINTER_NAME
# optional - defaulting to 127.0.0.1
listen_adress: 0.0.0.0
# needed - use above 1024 as nonroot
listen_port: 4750
#optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web
**#commands allowed to be used mid print
midprint_allowed:
command_1
command_2
#commands not reporting output to the klippy console
ack_needers:
command_1
command_2**
```
This is what I tested now: 
added G92 E0, to reset the extruder
`ack_needers = [ "G0", "G1", "G28", "G90", "G91", **"G92"**, "M0", "M24", "M25", "M83", "M84", "M104", "M112", "M117", "M140", "M141", "DUMP_TMC", "FIRMWARE_RESTART" "", "SET_PIN", "STEPPER_BUZZ" ]`

And in midprint_allow, 
AFAIK, M290 Zxx.xx, for babystepping the Z axis offset, 
needs a macro to work on klipper.
```
[gcode_macro M290 Z]
variable_parameter_Z:-0.02 #should be changed to a variable user input
gcode:
SET_GCODE_OFFSET Z_ADJUST={Z} MOVE=1
```
and the command that Kevin uses instead is:
`SET_GCODE_OFFSET Z_ADJUST=[your offset step] MOVE=1`
Z_ADJUST ensure to make a relative move from previous position,
and MOVE=1 make it running right after a move is finished.

Also M290 allows in midprint could be used if added with a macro to also offsets x and y axis which in is not needed in midprint command, IMO.

More commands to the midprint_allow array:
Set fan speed: M106 S<value>
Turn fan off: M107
Emergency stop: M112
Get current position: M114
Get firmware version: M115
Set extruder temperature: M104 [T<index>] [S<temperature>]
Set acceleration: M204 S<value>
Select tool: T<index>

So it will look something like this:

```
midprint_allow = [ **'SET_GCODE_OFFSET Z_ADJUST'**, 'DUMP_TMC', 'G10', 'GET_POSITION', 'HELP', 'M0', **'M104'**, **'M106'**, **'M107'**, **'M112'**, **'M114'**, **'M115'**, 'M140', 'M24', 'M25', 'M106', **'M204'**, 'M220', 'M221', 'M290', 'M292', 'QUERY_FILAMENT_SENSOR', 'SET_TMC_CURRENT', 'SET_PIN',
				'SET_PRESSURE_ADVANCE', 'SET_VELOCITY_LIMIT', **'T'**,  ]

```
thanks for making something so great, and reviewing my changes.

Stephan3, 2019-11-08:
for:
SET_GCODE_OFFSET Z_ADJUST you could use the babystep buttons too. i added it anyways.

good findings. M112 wow , how could i miss that. Emergency button does in any condition.

I added as you proposed, Good Job.  pleas test latest master.

Stephan3, 2019-11-23:
I asume we are done.

closed 2019-11-23 by Stephan3