Stephan3/dwc2-for-klipper #31: DWC2 uses G-codes missing in Klipper

antst, 2019-11-06:
DWC2 uses some RRF-specific G-codes (especially M120/121 for push/pop, which wraps all macros in case of RRF), which are missing in Klipper.
Of course, there is always possibility to define bunch of empty/non-empty macros in Klipper. To stop DWC2 from complains, but I am unsure it is best course of action.
Would it be better if DWC python code would somehow take care of it?

Stephan3, 2019-11-08:
as i never used rrf for a long period, i am not sure what you mean.can you give me an example?

antst, 2019-11-08:
Try extrude some filament in dwc2-klipper, and you will see M130/M121 errors

Stephan3, 2019-11-08:
Yes i see but what is the ecxpected behavior?

antst, 2019-11-08:
You pop-ups with errors about wrong g-codes?
I’d say there should be none.
 dwc wraps up any embedded into it  macro in m120/m121 which are rrf-specific. and I expect that at least those g-codes has to be filtered out by your python code.

Stephan3, 2019-11-09:
Again what is a valid usecase in ref for m120 and m121? What does it do? Try to explain in easy words. There might be possibilities in Klipper

tonn0297, 2019-11-11:
M120: After this command endstops will be kept enabled when not homing
M121: After this command endstops will be kept disabled when not homing
that is the default for marlin.

i don't see a use in klipper for this as we don't print from sd.
And the error comes from klipper and not dwc.

Stephan3, 2019-11-11:
We print from virtual sd. Anyways I just can make it ignoring m120/m121

antst, 2019-11-11:
> Again what is a valid usecase in ref for m120 and m121? What does it do? Try to explain in easy words. There might be possibilities in Klipper

It is not use-case.
DWC2 JS code, as it comes from Duet, wraps all stuff in M120/M121 (Push the state of the RepRap machine onto a stack/Recover the last state pushed onto the stack), those are different from Marlin.

Therefore, if you heat up extruder, go to DWC2 webUI , dashboard and just extrude some amount of filament, you will get popup about wrong G-codes.

[image]

antst, 2019-11-11:
And it is not the same SAVE_GCODE_STATE/RESTORE_GCODE_STATE in Klipper, so there is no point to define them as aliases to those.

Therefore, it is always possible to define empty macro M120 and M121, of course, to stop those annoying complains. But taking into account that this is generic thing, which affects everyone, I assume that it is better to be fixed on your side in python code which emulates RRF.
Maybe I am wrong.

FHeilmann, 2019-11-13:
Coudl you elaborate which particular detail makes you say the two sets of gcodes are not equivalent?

Both can be used to save and restore things like feedrate, absolute/relative state as well as position of axes and extruder. 

Granted the way they are called is slightly different, but in general they should both accomplish the same thing here, which is making sure that the extrude move triggered by the UI does not modify the state of the printer or the extruder axis, and the state previous to the extrude move is restored after the material has been extruded.

antst, 2019-11-13:
M120/M121 does not store/restore positions, otherwise you would not wrap in store/restore macro which moves carriage, as result would be that carriage will get back to original position :)

FHeilmann, 2019-11-13:
Well you can use SAVE/RESTORE_GCODE_STATE without giving a cr** about the position it stores. 
The important bit is that SAVE_GCODE_STATE does all the things M120/M121 does in this case, plus some extra stuff, which can be ignored.

So I consider simply aliasing the two here a valid solution to the problem at hand, wouldn't you agree?

antst, 2019-11-15:
Nope.  It can not be ignored )
What will he result if 

SAVE_GCODE_STATE
G92
G0 X10
RESTORE_GCODE_STATE

?

FHeilmann, 2019-11-15:
what does that gcode snippet have to do with the button you're pushing in dwc2?!

antst, 2019-11-18:
You miss. As RESTORE_GCODE_STATE will restore X,Y,Z coordinates. While M120/M121 will not.
And DWC2 wraps motion of X,Y,Z in 120/121.

FHeilmann, 2019-11-19:
`RESTORE_GCODE_STATE` will only move back if `MOVE=1` is passed to it.

Stephan3, 2019-11-23:
@antst 
Current master includes a save/restore solution. I tested few things and it does. Please close this once you are okay with that.

antst, 2019-11-23:
I will test, but I don't see how this will be OK :) If you move X by 20mm wrapped with klipper's store/restore, you achieve not what you expect.

antst, 2019-11-23:
Ah, this is something I missed completely! That there is MOVE=0 option for restore. Then it does exactly equivalent of RRF restore!
Thanks

antst, 2019-11-23:
But then I am lost. I checked, and M120/M121 are in the code since september. 
And I have DWC updated from the master last time about a week ago. And i still have M120/M121 warnings. Apparently, it looks like I am hitting some other issue.

Stephan3, 2019-11-23:
This here works now with latest klipper git. For now i printed 4 objects and everything works okay. So i am going to close this.

closed 2019-11-23 by Stephan3