Stephan3/dwc2-for-klipper #33: Problem with manual stepper (case-sensitivity, no response?)

Spacemarine2018, 2019-11-10:
I posted this problem originally in the Klipper bugtracker.  I was advised to replace DWC2-for-Klipper with Octoprint which made the problem go away. So I guess it's a DWC2-for-Klipper Problem:

My config: SKR Pro Board from Bigtreetech with TMC5160 V1.2 from Bigtreetech. Raspberry Pi 4 B running Duet Web Control

I want to use a manual stepper for cutting filament, so I defined it as:

[manual_stepper cutter]
step_pin = PD15
dir_pin = PE7
enable_pin = !PA3
step_distance = .003125
velocity = 70
accel = 1200

When I issue "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10"
the response is "failed: MANUAL_STEPPERThe value 'CUTTER' is not valid for STEPPER"

It makes no difference if I issue CUTTER or cutter on the gcode console, the output is the same and always uppercase.

Then I changed cutter to CUTTER in the printer.cfg, so it is now:

[manual_stepper CUTTER]
step_pin = PD15
dir_pin = PE7
enable_pin = !PA3
step_distance = .003125
velocity = 70
accel = 1200

When I run "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10" the g-code console get's stuck, as it is waiting for a return message that never comes. Also there is no movement from the stepper.

All steps can be seen in the klippy.log, I did a fresh restart before the experiment to get a clean log that shows only the behaviour described above.

Source: https://github.com/KevinOConnor/klipper/issues/2146

Stephan3, 2019-11-11:
Can you try to:

echo "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10" > /tmp/printer

in addition try to uppercase cutter in your config and report back please.

Spacemarine2018, 2019-11-12:
> 
> 
> Can you try to:
> 
> echo "MANUAL_STEPPER STEPPER=CUTTER ENABLE=1 MOVE=10" > /tmp/printer

Sorry, I stopped using DWC2 after I got it wo work with octoprint

> 
> in addition try to uppercase cutter in your config and report back please.

I already did that as I stated in the original post. However the part " Also there is no movement from the stepper." might have been due to some other problems. But I definately got a stuck console.

closed 2019-11-13 by Stephan3