Stephan3/dwc2-for-klipper #34: Not displaying tool temperature and infinite loading on some gcodes

saezels, 2019-11-10:
Just installed Klipper with DWC2.

I really like the interface but I cant get the UI to show my tool temperature. 

I can set the tool temperature and it is displaying on the Printer-LCD just fine. 

Also there are some g-codes that make the UI load infinitely and do not execute (e.g. M105).

I ran Klipper with OctoPrint before without problems. 

[P.S. recognized reiner calmund in the config, greetings from lev ;) ]

Stephan3, 2019-11-10:
Can you stop Klipper, clear the log(rm /tmp/klippy.log), perform a few actions you described and post the log here?

Probably I broke few things last week.

pizzle-d, 2019-11-10:
Renaming the extruder0 to extruder should fix it in web_dwc2.py

Stephan3, 2019-11-11:
@saezels please try latest dwc-for-klipper git, it should work now
@pizzle-d thanks for the quickfind

saezels, 2019-11-11:
I tinkered too much trying to solve the issue, have to do a fresh install and will Post the log later, thank you

saezels, 2019-11-11:
Ok my printer runs again. Extruder works with latest git. Infinite loading is still a problem.
Tried "M105" and "ABORT" (after PROBE_CALIBRATE), you can see it in my log. 
I've noticed a strange IP in the log: 192.168.178.29 (my Printer is running on 192.168.178.41)

Homing , temperature control and others are just working fine. 

klippy.log :

`
Stats 862.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=11977 bytes_read=871391 bytes_retransmit=36 bytes_invalid=0 send_seq=1285 receive_seq=1285 retransmit_seq=17 srtt=0.005 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000365 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.7 pwm=0.000
Stats 863.6: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=11983 bytes_read=872458 bytes_retransmit=36 bytes_invalid=0 send_seq=1286 receive_seq=1286 retransmit_seq=17 srtt=0.005 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000364 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.4 pwm=0.000
Stats 864.6: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=11989 bytes_read=873496 bytes_retransmit=36 bytes_invalid=0 send_seq=1287 receive_seq=1287 retransmit_seq=17 srtt=0.005 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000364 heater_bed: target=0 temp=17.7 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.8 pwm=0.000
Stats 865.6: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12013 bytes_read=874544 bytes_retransmit=36 bytes_invalid=0 send_seq=1290 receive_seq=1290 retransmit_seq=17 srtt=0.005 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000364 heater_bed: target=0 temp=17.1 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.6 pwm=0.000
Stats 866.6: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12037 bytes_read=875606 bytes_retransmit=36 bytes_invalid=0 send_seq=1293 receive_seq=1293 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000365 heater_bed: target=0 temp=17.3 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.6 pwm=0.000
Stats 867.6: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12043 bytes_read=876658 bytes_retransmit=36 bytes_invalid=0 send_seq=1294 receive_seq=1294 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000365 heater_bed: target=0 temp=17.7 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.1 pwm=0.000
Stats 868.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12067 bytes_read=877707 bytes_retransmit=36 bytes_invalid=0 send_seq=1297 receive_seq=1297 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.4 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.8 pwm=0.000
M105
DWC2:89522.42 - 192.168.178.29 - GET - /rr_gcode?gcode=M105
M105
passed: M105
Stats 869.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12091 bytes_read=878769 bytes_retransmit=36 bytes_invalid=0 send_seq=1300 receive_seq=1300 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.6 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.6 pwm=0.000
Stats 870.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12115 bytes_read=879831 bytes_retransmit=36 bytes_invalid=0 send_seq=1303 receive_seq=1303 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.6 pwm=0.000
Stats 871.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12121 bytes_read=880855 bytes_retransmit=36 bytes_invalid=0 send_seq=1304 receive_seq=1304 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.5 pwm=0.000
Stats 872.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=12145 bytes_read=881917 bytes_retransmit=36 bytes_invalid=0 send_seq=1307 receive_seq=1307 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.7 pwm=0.000
Stats 873.6: gcodein=0 mcu: mcu_awake=0.057 mcu_task_avg=0.000132 mcu_task_stddev=0.000042 bytes_write=12169 bytes_read=882994 bytes_retransmit=36 bytes_invalid=0 send_seq=1310 receive_seq=1310 retransmit_seq=17 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000366 heater_bed: target=0 temp=17.2 pwm=0.000  print_time=790.492 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.4 pwm=0.000
[... cut ...] 4.002 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.1 pwm=0.000
Stats 947.7: gcodein=0 mcu: mcu_awake=0.056 mcu_task_avg=0.000133 mcu_task_stddev=0.000042 bytes_write=17114 bytes_read=961747 bytes_retransmit=36 bytes_invalid=0 send_seq=1602 receive_seq=1602 retransmit_seq=17 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000380 heater_bed: target=0 temp=17.3 pwm=0.000  print_time=904.002 buffer_time=0.000 print_stall=0 extruder: target=0 temp=18.3 pwm=0.000
`

Stephan3, 2019-11-13:
Do you see that log you posted? it is a wall of text. Even if i like its great square geometry is it not very readable. you can attach files here.

closed 2019-11-23 by Stephan3