Stephan3/dwc2-for-klipper #38: File list on LCD

rafaljot, 2019-11-18:
If you are using LCD there is a menu "SD Card". It lists files from folder /home/pi/sdcard
Unfortunately Klipper doesn't suport subfolders.

dwc2-for-klipper stores  all files in "gcodes" subfolder so I can't print uploaded files using LCD menu. 

I made little  workaround:
[virtual_sdcard]
path: /home/pi/sdcard/gcodes

`
#<-----><------>self.sdpath = self.configfile.getsection("virtual_sdcard").get("path", None)
<------><------>self.sdpath = config.get( 'sd_path', self.configfile.getsection("virtual_sdcard").get("path", None))
`

then:

[web_dwc2]
(...)
sd_path: /home/pi/sdcard

Stephan3, 2019-11-23:
I am willing to add this. As you did this allready, does dwc realize that printing started when starting from lcd?

Stephan3, 2019-12-18:
No answer since a month? you alive?

rafaljot, 2019-12-27:
Sorry,
Unfortunately not. Over and over It try to reconnect.
[image]

Stephan3, 2020-01-16:
Okay you are asking for "file list on LCD" and then you are talking about restarts.
What ever you did/doing it breaks thing. If i can help with code or take a look at logs, let me know.

closed 2020-01-16 by Stephan3

rafaljot, 2020-01-25:
Because this restart happens only during printing when I start it from LCD. 
Yep, I'll try to manage this issue.

Stephan3, 2020-01-25:
Then I need a loggile please

trevjonez, 2020-04-23:
both sdcard lcd listing and the constant errors when a print is started via LCD are resolved by the combination of
https://github.com/KevinOConnor/klipper/pull/2767 and [Stephan3/dwc2-for-klipper#65](https://github.com/Stephan3/dwc2-for-klipper/pull/65)