Stephan3/dwc2-for-klipper #40: 'web_dwc2'  is not a valid configuration section

AbomShep, 2019-12-29:
I updated Klipper today (main branch) and DWC2 couldn't connect. "OK, I thought, it's been a long time and many changes, lets see if DWC2 has changed" It had, so I updated it as well. Now Klipper errors out with the infamous
`Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 131, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 235, in check_unused_options
    section,))
Error: Section 'web_dwc2' is not a valid config section`
I made sure the link was still there, and did the sed commands to modify Klipper's file. The link and web_dwc2's permissions are good. Can't figure out what I'm doing wrong. I had this once before and it was an actual problem, so I'm logging this just in case.

AbomShep, 2019-12-31:
Just did a fresh install with the same result.

Stephan3, 2019-12-31:
Your file is missing or the link is miss pointing

AbomShep, 2019-12-31:
You'd think right? as shown above everything shows correctly.

Stephan3, 2019-12-31:
There is no other reason why a klippy instance is reporting invalid config section

closed 2019-12-31 by AbomShep