Stephan3/dwc2-for-klipper #41: bed_mesh error

shadrincev, 2020-01-13:
Good day clipper has been updated and now an error is displayed when connecting
```
DWC2 starting at: http://0.0.0.0:5000
Extruder max_extrude_ratio=0.266081
Extruder max_extrude_ratio=0.266081
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 131, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 237, in check_unused_options
    section,))
Error: Section 'bed_mesh' is not a valid config section
DWC2:37652.39 - 192.168.1.232 - GET - /rr_connect?password=reprap&time=2020-1-14T0%3A47%3A32
DWC2:37652.74 - 192.168.1.232 - GET - /rr_filelist?dir=0%3A%2Fsys&first=0
DWC2:37652.77 - 192.168.1.232 - GET - /rr_config
DWC2:37652.79 - 192.168.1.232 - GET - /rr_filelist?dir=0%3A%2Fmacros&first=0
```
then, if you release an update, how do I update?

Stephan3, 2020-01-13:
Please read the error message. Thanks

closed 2020-01-13 by Stephan3

shadrincev, 2020-01-14:
I'm sorry, it was my mistake. Thanks