Stephan3/dwc2-for-klipper #43: Filenames with spaces

ZhenyaKa, 2020-01-16:
Is it possible to solve that problem?

Klipper can't print these files. 
Each time need to check spaces and rename files.

Stephan3, 2020-01-22:
for me a upload script is doing that. let me look into that

ZhenyaKa, 2020-01-22:
Yes, it's works but I upload files with sync software via samba.
And files has spaces :(

Stephan3, 2020-01-22:
Uploading using webif handles that. The point is, that this is a klipper thing. 
Which means, that klipper needs to open files from its "sdcard" withount whitespaces. 

The Idea i have is to rename files when filelist is created. ( themoment you click "jobs"  in the webif

Stephan3, 2020-01-22:
fixed

closed 2020-01-22 by Stephan3

guestisp, 2020-04-29:
I'm still experiencing this bug, when directly printing from a slicer (Slic3r++)

```
DWC2:96067.04 - 10.123.234.2 - POST - /rr_upload?name=0:/gcodes/Ender-hotend_plate%20%20%281%29.gcode&time=2020-04-29T23:34:19
M32 Ender-hotend_plate  (1).gcode
Uncaught exception GET /rr_gcode?gcode=M32%20"Ender-hotend_plate%20%20%281%29.gcode" (10.123.234.2)
HTTPServerRequest(protocol='http', host='10.11.12.150:4750', method='GET', uri='/rr_gcode?gcode=M32%20"Ender-hotend_plate%20%20%281%29.gcode"', version='HTTP/1.1', remote_ip='10.123.234.2')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 227, in get
    self.repl_ = self.web_dwc2.rr_gcode( self )
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 543, in rr_gcode
    handover = func_(params)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1084, in cmd_M32
    raise 'gcodefile' + fullpath + ' not found'
TypeError: exceptions must be old-style classes or derived from BaseException, not str
DWC2:96067.36 - 10.123.234.2 - GET - /rr_gcode?gcode=M32%20"Ender-hotend_plate%20%20%281%29.gcode"
```