Stephan3/dwc2-for-klipper #44: AttributeError: BedMesh instance has no attribute 'calibrate'

Wireheadbe, 2020-01-16:
Not sure what's going on to be honest. Running a very recent Klipper. When I run a BED_MESH_CALIBRATE; it correctly probes and I see output to the dwc2 console.

However, when trying to generate a bed map; I get an internal error:

```
HTTPServerRequest(protocol='http', host='octopi.wirehead.be:8080', method='GET', uri='/rr_download?name=0%3A%2Fsys%2Fheightmap.csv', version='HTTP/1.1', remote_ip='192.168.100.14')
Traceback (most recent call last):
  File "/home/jeffrey/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/home/jeffrey/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/jeffrey/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/jeffrey/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/jeffrey/klipper/klippy/extras/web_dwc2.py", line 216, in get
    self.repl_ = self.web_dwc2.rr_download(self)
  File "/home/jeffrey/klipper/klippy/extras/web_dwc2.py", line 376, in rr_download
    repl_ = self.get_heigthmap()
  File "/home/jeffrey/klipper/klippy/extras/web_dwc2.py", line 1302, in get_heigthmap
    z_matrix = self.bed_mesh.calibrate.probed_z_table
AttributeError: BedMesh instance has no attribute 'calibrate'
DWC2:05382.78 - 192.168.100.14 - GET - /rr_download?name=0%3A%2Fsys%2Fheightmap.csv
```

Wireheadbe, 2020-01-16:
Klipper version is v0.8.0-259-gb3c3b613

Wireheadbe, 2020-01-16:
I guess this is the culprit: 
https://github.com/KevinOConnor/klipper/commit/5b20ac416c409bb500a7cfae533a57d263ad0bb1

Wireheadbe, 2020-01-16:
in dwc2-for-klipper/web_dwc2.py;
line 1302; should become:

> z_matrix = self.bed_mesh.bmc.probed_matrix

Wireheadbe, 2020-01-19:
Also mkdir ~/sdcard/sys is needed on debian buster / ubuntu

Stephan3, 2020-01-22:
> mkdir ~/sdcard/sys

wont hurt fixed that first

closed 2020-01-22 by Stephan3

Stephan3, 2020-01-22:
fixed mesh display in latest master

commit db5023d4 (2020-01-22): fix #44

adding mkdir ~/sdcard/sys
commit de501859 (2020-01-22): fixes #44 - adjusted to latest klipper changes.