Stephan3/dwc2-for-klipper #47: M600 macro

shadrincev, 2020-01-27:
Good afternoon. There was a problem with the m600 command.
The code for replacing the filament on the command is written in the file printer.cfg:
```
[gcode_macro M600]
gcode:
    SAVE_GCODE_STATE NAME=M600_State
    PAUSE_MACRO

[gcode_macro PAUSE_MACRO]
gcode:
    PAUSE
    PARK_MACRO

[gcode_macro PARK_MACRO]
default_parameter_X: 0
default_parameter_Y: 0
default_parameter_Z: 20
gcode:
    M117 PARKING..
    G92 E0
    G91
    M83
    G1 E-3.14 F1500
    G1 Z{Z}
    G90
    M82
    G1 X{X} Y{Y} F3000
    M117 Nozzle Parked

[gcode_macro RESUME_MACRO]
gcode:
    M117  RESUMING..
    RESTORE_GCODE_STATE NAME=M600_State MOVE=1
    CLEAR_PAUSE
```

Dwc gives an error printing is interrupted, not paused:

> Warning:  Unknown command:"M117"
> 27.01.2020, 22:09:19	M98 P"0:/macros/PAUSE_MACRO"
> Warning:  Unknown command:"PAUSE"
> Warning:  Unknown command:"M117"
> 27.01.2020, 22:08:55	M98 P"0:/macros/RESUME_MACRO"
> Warning:  Unknown command:"M117"
> Warning:  Unknown command:"CLEAR_PAUSE"
> 27.01.2020, 22:07:10	Warning:  Unknown command:"M117"
> Warning:  Move exceeds maximum extrusion (1.302mm^2 vs 0.640mm^2)
> Warning:  See the 'max_extrude_cross_section' config option for details
> Error:  Move exceeds maximum extrusion (1.302mm^2 vs 0.640mm^2)
> 27.01.2020, 22:07:04	Warning:  Unknown command:"PAUSE"
> Warning:  Unknown command:"M117"
> 

Incomprehensible error
**Warning:  Move exceeds maximum extrusion (1.302mm^2 vs 0.640mm^2)** I have is worth:
**max_extrude_only_distance: 1000.0**

[a similar problem has arisen here,](https://github.com/KevinOConnor/klipper/issues/2413) there was advised to contact you.

Stephan3, 2020-01-29:
> Warning: Move exceeds maximum extrusion (1.302mm^2 vs 0.640mm^2) I have is worth:
max_extrude_only_distance: 1000.0

Thats not dwc related

> Warning: Unknown command:"M117"

This is a klipper message. dwc is just showing it. Can your try to echo that into /emp/printer and see what happens?

echo "M117 whatever" > /tmp/printer

read pause/resume macros autocalled in readme please

shadrincev, 2020-01-29:
I didn't quite understand what you meant. I will describe the order.
I start the printer and print the file. the M600 command is inserted in it when you need to change the filament:
CFFFP_20mm_cube_soft_edges.gcod
an error occurs.
I attach the log file:
klippy.log
a shortcut in the temp folder throws an error:
Безымянный.jpg, Безымянный1.jpg
here are the messages from the dwc window:
1234.txt
[Desktop.zip](https://github.com/Stephan3/dwc2-for-klipper/files/4128017/Desktop.zip)

Stephan3, 2020-01-29:
oh wow. please login with ssh to your computer(klipper is running on) and do:

echo "M117 whatever" > /tmp/printer

i assume that this wont work.

shadrincev, 2020-01-29:
I beg your pardon...
[image]
in the dwc console: 
29.01.2020, 21:10:38 | Warning:  Unknown command:"M117"

Then another, the extrusion button is not active, although the temperature is normal. Recently updated.
[image]

Stephan3, 2020-01-29:
please dont mix issues. This conversation confusing enough for you and me. 

So after M117, is that "whatever" message displayed on your display?

shadrincev, 2020-01-30:
> please dont mix issues.

Sorry, I'll create a separate request later. 

> So after M117, is that "whatever" message displayed on your display?

I don't have a screen, I only use DWC and the browser.

Stephan3, 2020-01-30:
Oh wow. If there is no screen, m117 command is not supported and therefore a unknown command for Klipper.
So that is Not an issue, that’s a expected behavior.

Can you please post your configuration here?

shadrincev, 2020-01-30:
The configuration is in the log file, but I duplicate it.

[printer.zip](https://github.com/Stephan3/dwc2-for-klipper/files/4132406/printer.zip)

 It's not about displaying the m117 message, not executing the pause script.

schnello, 2020-03-09:
The response section is missing:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#send-message-respond-to-host

Stephan3, 2020-03-09:
Let me finally finish this, as i dont know if you really got whats happening here.

Your M600 Macro calls your PAUS_MACRO.

Your PAUSE_MACRO calls your PARK_MACRO.

Your PARK_MACRO comtains a M117 command.

M117 is for display messages. As you did not installaded/configured a display M117 is a unknown command. This is what happens in klipper and its ecxpected. You probably never noticed as octoprint would not highlight it.

I will close this now

closed 2020-03-09 by Stephan3