Stephan3/dwc2-for-klipper #48: Height Map co-ordinates do not match bed width

lenne0815, 2020-01-29:
Sorry to reopen the issue like this but i did some further testing to see where it originates and came up with this:

having the same problem, in the csv there are 10 x 10 points, i only set up 4 x 4
heightmap.zip
Max z Probed point is 74.5 in the gui, even though the csv says 224.98

[Heightmap](https://github.com/Stephan3/dwc2-for-klipper/files/3964869/heightmap.zip)

Ah, i think the spacing is calculated wrongly, the points are much further apart then 21.498 it displays 10 points but i only set up 4, thats probably why.

Stephan3, 2020-01-29:
Hi! 

your CSV contains a 4x4 probing. dont know if i got your question right i cant see there 10 points.

[image]

https://github.com/Stephan3/dwc2-for-klipper/blob/476580121bf5ee7f8a45caff7f5f4df609253ac6/web_dwc2.py#L1312

The formula seems right, i need to check the values klipper is providing.

lenne0815, 2020-01-29:
Hey stephan, thanks for answering !

My klipper config gives 4x4 points like this:

probe_count: 4,4

Somewhere along the way it ends up being 10 points in the CSV, so the gui messes up the total values.

I dwc2klipper doesnt get the correct values from the klipper config.

Stephan3, 2020-01-29:
it get no values from config it get values from klippy.

Stephan3, 2020-01-29:
@lenne0815  can you please probe your bed 4x4 SAVE_CONFIG, and same with 6x6?
Aufter SAVE config i need each printer.cfg, so i can look into that.
Thanks

Stephan3, 2020-03-09:
no reply für 5 weeks -> close

closed 2020-03-09 by Stephan3