Stephan3/dwc2-for-klipper #49: Macro loading is broken on latest DWC releases

altvnk, 2020-01-30:
After updating to DWC 2.0.6 i have noticed that macros are not being loaded anymore. Just checked recently released 2.0.7 and it's affected too.
<details>
  <summary>Log message</summary>

```
Uncaught exception GET /rr_filelist?dir=0%3A%2Fmacros&first=0 (192.168.145.121)
HTTPServerRequest(protocol='http', host='raspberrypi.local:4750', method='GET', uri='/rr_filelist?dir=0%3A%2Fmacros&first=0', version='HTTP/1.1', remote_ip='192.168.145.121')
Traceback (most recent call last):
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/tornado/gen.py", line 326, in wrapper
    yielded = next(result)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 218, in get
    self.repl_ = self.web_dwc2.rr_filelist(self)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 454, in rr_filelist
    "date": datetime.datetime.fromtimestamp(os.stat(el_path).st_mtime).strftime("%Y-%m-%dT%H:%M:%S")
UnboundLocalError: local variable 'el_path' referenced before assignment
DWC2:17503.64 - 192.168.145.121 - GET - /rr_filelist?dir=0%3A%2Fmacros&first=0
```

</details>

shadrincev, 2020-01-31:
Suddenly you have the same as me. I updated DWC to version 2.0.6 and I lost the possibility of manual extrusion, but I put the DuetWebControl-**SBC** version.zip, and now updated to 2.0.7 and put it DuetWebControl-**SD**.zip. Everything is back. It might help...

altvnk, 2020-01-31:
Manual extrusion was broken for me too, resolved by clearing browser cache.

Stephan3, 2020-02-01:
https://github.com/Stephan3/dwc2-for-klipper/commit/997655aa4ef3e70b9ca1aba47a7560d42076028a

this is solved since 2?3? days

closed 2020-02-01 by Stephan3

altvnk, 2020-02-01:
Strange, I pulled master before submitting the issue. 
Thanks, I will double check.

altvnk, 2020-02-02:
So, as I said I'm on latest master, on commit you have referenced
`git log | head -n 1
commit 997655aa4ef3e70b9ca1aba47a7560d42076028a`
Symlink is in place, klipper restarted and everything, but I still getting 500's. Any ideas what else could be wrong?

Wireheadbe, 2020-02-04:
Same issue here

Wireheadbe, 2020-02-04:
Line 454 should be replaced:
#"date": datetime.datetime.fromtimestamp(os.stat(el_path).st_mtime).strftime("%Y-%m-%dT%H:%M:%S")
"date": datetime.datetime.fromtimestamp(os.stat(self.klipper_config).st_mtime).strftime("%Y-%m-%dT%H:%M:%S")

Stephan3, 2020-02-04:
Yes so klipper macros get the date of the config file. that looks right to me, i fixed it.

Wireheadbe, 2020-02-04:
Yep - seems like the most logical thing to do, thanks for committing :)

altvnk, 2020-02-06:
That is definitely works as intended now. Many thanks!