Stephan3/dwc2-for-klipper #51: Page Not Found After Adding callback for gcode responses

NedalLive, 2020-02-29:
Hello,
After installation I had access to the web page, however it did not seem to communicate with Klipper.
[image]

I therefore carry out the modification operation of the gcode.py file by hand, after this modification it is impossible to find the web page ...
I tried to redo an installation from the beginning without success
[image]

Stephan3, 2020-02-29:
Please provide a log.

NedalLive, 2020-02-29:
my problem comes from editing the file by hand with Notepad ++ it added tabulation etrage ...
Corrected in ssh is the problem is solved

closed 2020-03-09 by Stephan3