Stephan3/dwc2-for-klipper #52: "RETRACT" and "EXTRUDE" buttons disabled

tealbrains, 2020-03-12:
Hi Stephan,

Thanks for the excellent work. I have just installed yesterday klipper +dwc2 on an Orange Pi Zero Plus and it is working so far with the usual "beta" issues.

I know that this was brought up in #7  , but I am not sure if that was solved or if it just died because the OP never reported back.

I think this issue is purely related to dwc2 for the following reasons:
1) R&E buttons worked this morning
2) R&E buttons no longer worked this afternoon
3) Tried restarting klipper, the printer. Nothing worked.
4) After messing with the html code of the buttons poking styles ( "disabled" and v-btn-disabled I think it was), and then refreshing the browser brought the buttons back, and they work again as expected.

If you need more info let me know. Happy to help.
I don't think it is a coincidence but I could be wrong.

tealbrains, 2020-03-12:
If there is a function that looks at the temperature of the extruder to block these that would be my suggestion. The dashboard was showing the right value, but maybe it gets corrupted somewhere?

Stephan3, 2020-03-23:
@tealbrains 

1. & 2. once webif restarts and or had state "off"(klipper shutdown/mcu failure) the webif needs reload to make buttons work. this has nothing to do with the translator or i did not find the issue.
So if you findout which data is missing during "OFF" i can add that.
During Off i provide rr_status0 i kept it readable for humans:
https://github.com/Stephan3/dwc2-for-klipper/blob/05f75e86b7fcb59ce7d11024eab449cce373c084/web_dwc2.py#L594

3. restarting klipper works. RESTART and FIRMWARE_RESTART works just as ecxpected

tealbrains, 2020-03-24:
@Stephan3 Could this have anything to do with something I read about gcode execution acknowledgement? Today I observed a different behaviour where I pressed "EXTRUDE" once and it worked, but then it kept spinning so I was unable to click again. This could be a different problem to the first post but could be related? I will try running RESTART and FIRMWARE_RESTART to see if that solves the problem and report. Reloading the website solves this issue I just mentioned though, so not sure it's an mcu issue.

killisch, 2020-03-29:
I've got the same issue using @blackstumps klipper image for the replicape. not sure how to get it to show the buttons. but the extruder is controlled correctly from a gcode file and i can start a print.

Stephan3, 2020-04-08:
You can add things to the ack needers and chack. Cant tell out the top of my head what commands this botton sends. Try to find a elegant way to send a ack to that button.

https://github.com/Stephan3/dwc2-for-klipper/blob/f2b8ebbc72896c216bb26029c7409ae4edbe4933/web_dwc2.py#L1184

reefwarrior, 2020-05-01:
Hi! I seem to have the same problem of not being able to extrude through the web interface :( Any help/update with regards to this? (Both extrude and retract are greyed out)

Stephan3, 2020-05-02:
Try what stated one post above

reefwarrior, 2020-05-02:
> Try what stated one post above

Brilliant, thank you, it worked. I saw there was a misplaced '""' or something in the line - took the 'wrong' one(s) out, and it works! :)

Stephan3, 2020-05-03:
indeed there is one :D thanks for pointing it out -> fixed

commit 0e0ab62f (2020-05-03): Update web_dwc2.py

fixes #52
closed 2020-05-03 by Stephan3