Stephan3/dwc2-for-klipper #53: M80/81 psu control

shadrincev, 2020-03-16:
Good afternoon. How can I activate the enable / disable button? I write a macro for the M80 and M81 commands that are sent when it is clicked, but an error occurs: Malformed command 'RESPOND type:command MSG=action:poweron'. Or is this also not implemented?

```
[respond]
default_type: echo

[gcode_macro M80]
gcode: RESPOND type:command MSG=action:poweron

# power off
[gcode_macro M81]
gcode: RESPOND type:command MSG=action:poweroff
```

Stephan3, 2020-03-23:
I dont have a setup to test this. If someone is willing to add this, i am happy with it. Thanks

shadrincev, 2020-03-23:
I'm willing to help, but I don't know how. If you specify what you need to implement, I will do it.

alligator77, 2020-04-24:
Klipper is supported for controlling logic on pins; create M80 / M81 macros yourself.
I use SKR 1.4 and Lerdge power module, this macros works fine:
```
[gcode_macro M81]
gcode:
      SET_PIN PIN=ShutDownPrinter VALUE=1

[output_pin ShutDownPrinter]
pin: P1.0
```

Stephan3, 2020-05-02:
So as the Solution is found, I close this

closed 2020-05-02 by Stephan3