Stephan3/dwc2-for-klipper #54: Connectivity issues

tealbrains, 2020-03-18:
Description: 
- Klipper is running
- DWC2 is running

1) The dashboard shows 2 extruders (my printer only has one), and wrong data overall as if it had not been able to connect to the printer. Looks like default information when the communication is not established.
2) The printer is listed under /dev/serial/device-by-id/.....
3) If I stop klipper and restart it, sometimes the dashboard will show the right printer data (one extruder, right temperatures, etc).
4) Sometimes I have to go as far as restarting the OrangePi Zero.

Not sure where this is happening but if you point me in the right direction I will get whatever log files or anything needed to debug it. It could be a klipper problem maybe?

Stephan3, 2020-03-23:
I never saw a behavior like this. You can try run klipper from the console. 
If you run klippy from console(not as a daemon) it shows you whats happening.

tealbrains, 2020-03-24:
I am going to try running FIRMWARE_RESTART next time it happens and will report, I did not attempt this yet. It would be maybe handy to have this as a button? Maybe less people will report issues if the button solves the problem.

Stephan3, 2020-04-04:
Well as i do this often and it works for me i close this. Seems fine as you never reported back. Thanks

closed 2020-04-04 by Stephan3

tealbrains, 2020-04-05:
I was facing major issues two days ago and decided to address this problem. I have a good USB cable (ferrite bead choke) but it is 1m long which could have been the problem although a bit strange. I have recompiled the firmware and flashed the board (atmega2560 based) to have Serial running with a baudrate of 115200 instead of 250000 (default) and changed the parameter in the config file of Klipper as well.
The linux box seems to be able to detect the printer with no issues so far.