Stephan3/dwc2-for-klipper #56: Starting next job

tealbrains, 2020-03-22:
Hi,

I think there is a little when starting a new job after finishing one. It takes a couple of attempts. The first one will execute the end of the previous job again (it will not load the new job and will display the old one as 100%), then once that is done you have to reload the new job for it to start. That's what I have observed. It's not a major issue as there is a way around it but something I thought was worth mentioning.

Stephan3, 2020-03-23:
I dont know what you are reffering to. For me i start / stop / pause /cancel Jobs daily. i cant reproduce this. maybe your hardware is just very slow?

closed 2020-03-23 by Stephan3

tealbrains, 2020-03-24:
I can try to take a video next time. I can confirm it's not just a one time thing, it's just happened again today. It may not be anything to do with your dwc2 port, maybe it's a klipper thing. It seems as if the previous job had not been fully cleared.