Stephan3/dwc2-for-klipper #60: After installation I have errors in log file

Kotletowsky, 2020-04-04:
[image]
And I can't start dwc. Webmenu is loaded, but still  blinking. And get message: o.temps.state is undefined

Stephan3, 2020-04-04:
Please try to read the text of the error message, thanks.

Kotletowsky, 2020-04-04:
But I have value in z_offset. I had DWC2 working before new installation, printer.cfg it's the same

Stephan3, 2020-04-04:
The config parser is telling a different story. This has nothing to do with dwc.

Kotletowsky, 2020-04-05:
I change offser from -1 to 0 and it's work. 
But there is anoter problem. I want to execute firmware_restart from console and I get:
JSON.parse: unexpected non-whitespace character after JSON data at line 1 column 22 of the JSON data

Stephan3, 2020-04-05:
provide the log. cant answer to that

Kotletowsky, 2020-05-01:
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper/files/4563023/klippy.log)
still getting json parse errors

rafaljot, 2020-05-01:
and config, pls.

rafaljot, 2020-05-01:
You need to clean up the config.
The last section [gcode_macro end_g] is mixed up with [web_dwc2], [virtual_sdcard] etc.

Kotletowsky, 2020-05-01:
json problems moved to another topoic

closed 2020-05-01 by Kotletowsky