Stephan3/dwc2-for-klipper #64: Which DWC2 version should I use?

Tomblarom, 2020-04-14:
Hey Stephan! Could you please clarify which version is recommended to use? The `README.md` says "2.0.6/DuetWebControl-SBC.zip". The [official klipper docs](https://klipper.info/klipper-+-dwc2-1/installing-dwc2) are using "2.0.0-RC5/DuetWebControl.zip". Can I safely use the latest version of DWC2?

Latest version by now is 2.1.1: https://github.com/chrishamm/DuetWebControl/releases

Stephan3, 2020-04-14:
I haven’t upgraded for a long time as my printers are always on. Probably it will just work with 2.1.1

Stephan3, 2020-04-14:
Pls test and report back

trevjonez, 2020-04-18:
I just updated from RC5 to 2.1.2 this week on my three machines. It mostly just works. There is a handful of issues but I have not paid close enough attention to know where at what to file since they have not been anything that can't be worked around.

Tomblarom, 2020-04-19:
Since I'm not using any Duet board, I can't give official validatable feedback. Using SKR 1.4 + Klipper was curious if this works. Still testing.

Torreip, 2020-04-20:
@Tomblarom you do not need duet hardware. Working for me with 2.1.1 + SKR 1.1 + current master branch of klipper.

Stephan3, 2020-05-03:
Updated to 2.1.6 two days ago looks fine, readme updated.

closed 2020-05-03 by Stephan3