Stephan3/dwc2-for-klipper #66: webif issue

Ahtyng85, 2020-04-29:
hello all!
my DWC2 doesn't start after one FIRMWARE_RESTART. I can't understand what happened and how to fix it. The part of last log:

```
[virtual_sdcard]
path = /home/pi/sdcard

[web_dwc2]
printer_name = Reiner Calmund
listen_adress = 0.0.0.0
listen_port = 4750
web_path = dwc2/web
=======================
DWC2 failed to start, no webif found in /home/pi/sdcard/dwc2/web
Extruder max_extrude_ratio=0.266081
Starting serial connect
got {'#receive_time': 21393.3384946, u'next_clock': 604537344, u'oid': 8, u'value': 32740, '#name': u'analog_in_state', '#sent_time': 21393.285494006}
got {'#receive_time': 21393.428468542, u'next_clock': 611017344, u'oid': 17, u'value': 32748, '#name': u'analog_in_state', '#sent_time': 21393.391318706}
Loaded MCU 'mcu' 77 commands (v0.8.0-433-g9730be8e-dirty-20200428_213816-debianPC / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-12+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c2=PB10,PB11 STEP_DELAY=2 INITIAL_PINS=!PC13 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 CLOCK_FREQ=72000000 BUS_PINS_spi1=PA6,PA7,PA5 RESERVE_PINS_USB=PA11,PA12 MCU=stm32f103xe STATS$
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder

 
```

exxperto, 2020-04-29:
I think this a reason  `DWC2 failed to start, no webif found in /home/pi/sdcard/dwc2/web`
Just put and unzip DuetWebControl-SD.zip

Stephan3, 2020-05-02:
Read install instructions.

closed 2020-05-02 by Stephan3