Stephan3/dwc2-for-klipper #69: internal server error 500

Kotletowsky, 2020-05-01:
Every time, when I want to send cmd by console, I get internal server error 500 message.
Especially, when I click something in Z babystepping.
How to solve that ?

rafaljot, 2020-05-01:
log pls.

ps: zapraszamy na reprapy.pl

Kotletowsky, 2020-05-01:
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper/files/4563586/klippy.log)

[printer_cfg.txt](https://github.com/Stephan3/dwc2-for-klipper/files/4563599/printer_cfg.txt)

rafaljot, 2020-05-01:
1. As I said in before, clean up the config first. Remove this line and all after 
`#*# <---------------------- SAVE_CONFIG ---------------------->
`
Don't write any configuration manually after this line in the future.

2. I'm not sure you can use comments in this way in macro.

```
[gcode_macro start_g]
gcode:
   G28
#   BED_MESH_CLEAR
#   BED_MESH_PROFILE LOAD=ENDER3

```
Delete these comments please.

Kotletowsky, 2020-05-02:
deleted, cleaned. Still, when I want to use babystep: 
[image]

rafaljot, 2020-05-02:
try the same using Octoprint. We'll find out if the problem is with DWC2 or Klipper.
It works for me:
[image]

Stephan3, 2020-05-03:
@Kotletowsky 
You logfile dos not contain a M290 . If we want to analyse your problem, we need a valid log here, thanks.

brycedjohnson, 2020-05-06:
Saw the same error when trying to set the hot bed.  New dwc2 install coming from an octoprint/octopi. 
[image]
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper/files/4588619/klippy.log)

brycedjohnson, 2020-05-06:
My issue went away by reverting to older klipper, the same as this one: https://github.com/Stephan3/dwc2-for-klipper/issues/72#issuecomment-624814547

closed 2020-05-25 by Stephan3