Stephan3/dwc2-for-klipper #71: adding an option in config - to make DWC2 connect multiple printers

manu7irl, 2020-05-05:
Hi Stephan, I have been experimenting, the making of a Klipper server on PC hardware with multiple printers conected to it, and I got multisession klipper install working flawlessly.
In order to get also multiple DWC2 session working I need it to connect to multiple virtual serial endpoints:
klippy.py connect by default to /tmp/printer, but it can be ran with a specified virtual serial endpoint via using flag: 
```-I /tmp/your-printer-name```
which is grant a way to implement multiple klipper sesions.

By reading your web_dwc2.py, I can see that you use explicitly the default endpoint name, in this line:
```self.serial = serial.Serial( '/tmp/printer', 250000, timeout=0.050 )```
and in order to get it working with any custom endpoints name, I propose to add this line instead: 
```self.serial = serial.Serial( config.get("serial_path", "/tmp/printer" ), 250000, timeout=0.050 )```
with this line, we can add in [web_dwc2] one more option:
```serial_path: /tmp/my-printer-name```
but is is not mandatory as the serial_path will be defaulted to /tmp/printer if not specifed.
Could you please take the time to add this?
Or please consider parsing that serial endpoint name from klippy.py response, so it will not be needed to bother about it... 
I do not have enough python skills to found what to parse from klippy.py...
thanks a lot, for your awesome work!
Manu7irl.

Stephan3, 2020-06-30:
its not planned yet. Sorry.

closed 2020-06-30 by Stephan3

manu7irl, 2020-06-30:
Ok thanks I got it working for 2 months now no issues...