Stephan3/dwc2-for-klipper #72: Error:  'dict' object has no attribute 'get_float'

brycedjohnson, 2020-05-06:
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper/files/4588574/klippy.log)

New dwc2 install coming from an octoprint/octopi.  Got that error when adjusting speed factor

[image]

alfrix, 2020-05-06:
Yeah i just updated and yes, Klipper changed a lot stuff 
since i'm not in the mood of debugging i just reverted back to May 4

if you need help:

```
sudo systemctl stop klipper
cd ~/klipper
git reset --hard 9a92b346fd81386c54464b45d8310dab972b6bbd
make clean
make 
make flash FLASH_DEVICE=yourdevicename
sudo systemctl start klipper
```

brycedjohnson, 2020-05-06:
Thanks @alfrix that appeared to fix it.  I'll leave the issue open for now unless someone wants me to close it?

th33xitus, 2020-05-08:
+1 on that issue.
Something is broken when updating to the newest commit of klipper.

Sending a heater command via DWC2 to my printer gives me the following error:
```
G10 P0 S120
Operation failed (Reason: 500: Internal Server Error)
```
What's really weird is, a G10 is usually a retract, so something is really broken when issuing a heating command.

Heating the bed give another error:
```
M140 P0 S50
Operation failed (Reason: 500: Internal Server Error)
```

Homing does work, but when moving the axes manually via the buttons in the interface the following error messages occur:
`Error: 'dict' object has no attribute 'get_commandline'`
`Error: 'dict' object has no attribute '_params'`

manu7irl, 2020-05-08:
Try to upgrade the dwc2 server files from duet web control website.
I am on latest one everything is ok.

BlackStump, 2020-05-08:
G10 with Reprap Firmware for which Duet Web Control is design for is for Temperature Control,
You need to remember Duet Web Control was never design for klipper.
fyi @th33xitus 
https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting

pluuuk, 2020-05-12:
#75 might fix this too.

manu7irl, 2020-05-12:
> G10 with Reprap Firmware for which Duet Web Control is design for is for Temperature Control,
> You need to remember Duet Web Control was never design for klipper.
> fyi @th33xitus
> https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting

Hi man I made a new patch for your script.
It is in my forked folder.
https://github.com/manu7irl/dwc2-for-klipper
You could have a look at it.

BlackStump, 2020-05-12:
Thanks, I have done the update

manu7irl, 2020-05-12:
all in all @pluuuk got so far that the need for a patch is obsolete!
I will test that in the next coming hours

Stephan3, 2020-06-30:
fixed

closed 2020-06-30 by Stephan3