Stephan3/dwc2-for-klipper #73: Printer instance has no attribute 'try_load_module'

markwinger, 2020-05-09:
I have been trying to install dwc2 for 2 days.  I was running octoprint with klipper and tried installing as described at:

https://github.com/Stephan3/dwc2-for-klipper

but that fails at wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.6/DuetWebControl-SBC.zip
 with the error indicating 2.1.6 does not exist.  I tried using:
wget https://github.com/chrishamm/DuetWebControl/releases/download/2.0.0-RC5/DuetWebControl.zip

which is described at:https://klipper.info/klipper-+-dwc2-1/installing-dwc2

That installed but the gui would not come up.  If I open octoprint the klipper  console "status" says:

Internal error during connect: Printer instance has no attribute 'try_load_module'

I even tried installing on a clean raspbian system with the installer.sh  descibed at:https://github.com/manu7irl/klipper-DWC2-installer?files=1.
I only have one printer so this is more that I need but I tried it anyway.  
With this I could not get dwc2, or octoprint guis to open and of course klipper then is not running.

I did install this a few months ago and it worked but I decided not to switch then so went back to octoprint. (no longer installed).

**The dwc2 gui fails to open(Unable to connect), and klipper does not start with this try_load_module error.**  

What next????

manu7irl, 2020-05-09:
The problem is related to the big klipper commit that has been done a few
days ago.
It made some changes to the gcode.py file with that the change that Stephan
has made in that file to make dwc2 working are not inline with the new file
structure.
We need to revert to an older commit and wait for Stephan to make the
changes needed in order to get that fixed.

markwinger, 2020-05-09:
On github 2.1.7 it showed a duetwebcontrol-SD.zip so I tried that too.  No difference.

Stephan3, 2020-05-09:
@manu7irl you pointed the Klipper commit allready. I have not the time doing much theese days.

manu7irl, 2020-05-09:
I will try to make the changes needed to your sed commands. I will try to
send a pr.

markwinger, 2020-05-09:
Octoprint works with klipper if I comment out the [web_dwc2] section.

BlackStump, 2020-05-09:
If you are desperate to try something other then Octoprint you could try 
[Mainsail](https://github.com/meteyou/mainsail)
It is alpha but seems to be quite functional, it has a DWC2 look about it.

pajak15, 2020-05-10:
I had the same issue today while doing a fresh install. Try changing at line 49 in the web_dwc2.py "try_load_module" with "load_object". As stated before there was a commit in klipper where they renamed try_load_module() to load_object().

pluuuk, 2020-05-10:
a simple sed command to replace try_load_module with load_object works.

Szeker, 2020-05-10:
@pluuuk: Are you on the latest Klipper? Simply replacing the "try_load..." to "load_object" seems not enough. "gcode.py" also has been modified in the latest version, so the original patch probably needs to be updated also.

pluuuk, 2020-05-10:
@Szeker , I opened #74 with the fixes to get dwc2 working again.

markwinger, 2020-05-10:
I am away now but i will try the fix tomorrow and update.

Get BlueMail for Android<http://www.bluemail.me/r?b=15860>

markwinger, 2020-05-11:
OK, one install says to use wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.6/DuetWebControl-SBC.zip.  But this does not exist, I get 2020-05-11 06:55:38 ERROR 404: Not Found.

The other: wget https://github.com/chrishamm/DuetWebControl/releases/download/2.0.0-RC5/DuetWebControl.zip is well over a year old.  

There are no SBC files after 2.1.4.  What do you recommend?

manu7irl, 2020-05-11:
```wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.7/DuetWebControl-SD.zip```

markwinger, 2020-05-11:
OK, that was the last one I pulled.  So just to clarify (I don't understand git) if I just repeat the install process these fixes will get installed?

manu7irl, 2020-05-11:
you need to clone @pluuuk folder instaed of Stephan because this work is not merged yet...
git clone https://github.com/pluuuk/dwc2-for-klipper

markwinger, 2020-05-11:
Ok, still no joy.

I used the instructions on the page you pointed to, substituting pluuuk for Stephan3 on the clode of dwc2-for-klipper.git.  The install seems to work.  Octoprint no longer complains about try_load_module, but when I open ip:4750, no gui.

manu7irl, 2020-05-11:
Use my script and report there I got it patched, AFAIK, it should work now.
https://github.com/manu7irl/klipper-dwc2-installer

markwinger, 2020-05-11:
I just tried to install dwc2 with it and it complains that the PrinterFarm direcoty does not exist.  Do I need to unistall everything and install everything with you script to move it all to the PrinterFarm?

pluuuk, 2020-05-11:
@markwinger, did you configure the virtual_sdcard and patched gcode.py using the sed commands?

pluuuk, 2020-05-11:
@manu7irl, the dwc2-for-klipper repo is 7 months old. I'm not sure if @markwinger will be able to use your installer.

pluuuk, 2020-05-12:
These are the ammended instructions to run the fork that fixes these issues:
`sudo systemctl stop klipper
cd ~
mv klipper klipper_backup 
PYTHONDIR="${HOME}/klippy-env"
virtualenv ${PYTHONDIR}
${PYTHONDIR}/bin/pip install tornado==5.1.1

git clone https://github.com/KevinOConnor/klipper.git
git clone https://github.com/pluuuk/dwc2-for-klipper.git
ln -s ~/dwc2-for-klipper/web_dwc2.py ~/klipper/klippy/extras/web_dwc2.py

mkdir -p ~/sdcard/dwc2/web
mkdir -p ~/sdcard/sys
cd ~/sdcard/dwc2/web 
wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.6/DuetWebControl-SBC.zip
unzip *.zip && for f_ in $(find . | grep '.gz');do gunzip ${f_};done
sudo systemctl start klipper`

You will still need to configure the virtual_sdcard as in the readme

markwinger, 2020-05-12:
Yes:
[virtual_sdcard]
path: ~/.octoprint/uploads/

and yes
gcode=$(sed 's/self.bytes_read = 0/self.bytes_read = 0\n        self.respond_callbacks = []/g' klipper/klippy/gcode.py)
gcode=$(echo "$gcode" | sed 's/# Response handling/def register_respond_callback(self, callback):\n        self.respond_callbacks.append(callback)/')
gcode=$(echo "$gcode" | sed 's/os.write(self.fd, msg+"\\n")/os.write(self.fd, msg+"\\n")\n            for callback in self.respond_callbacks:\n                callback(msg+"\\n")/')
echo "$gcode" > klipper/klippy/gcode.py

What about PrinterFarm directory?  Do I have to uninstall klipper and resinstall it with your script?  Klipper is installed and running with octoprint and I would like to keep it and add dwc2 so I can use either as I did when I tried it a few months ago.  Can that be done with you installer.sh script?

pluuuk, 2020-05-12:
@markwinger , please see my last comment, I changed some things in the fork to remove the need to run the sed commands to modify gcode.py. This should allow you to install this much easier. 
I'm not familiar with @manu7irl install scripts.

markwinger, 2020-05-12:
This still fails:
wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.6/DuetWebControl-SBC.zip

I changed it to 2.1.7/DuetWebControl-SD.zip.

Installed ok, but klipper fails to start again with a new error as shown in octoprints klipper console:
Internal error during connect: GCodeParser instance has no attribute 'register_respond_callback'

pluuuk, 2020-05-12:
make sure you're running a clean klipper install, it seems you're running a
modified gcode.py
```
cd klipper
git checkout -- klippy/gcode.py
```

Should help you clean it up.

markwinger, 2020-05-12:
No change.  So I deleted gcode.py, then did checkout again.  Still no change.

pluuuk, 2020-05-12:
```
  cd dwc2-for-klipper
  git checkout -- .
  git pull
````

markwinger, 2020-05-12:
Now klipper starts but the dwc2 gui will not connect.

pluuuk, 2020-05-12:
Can you attach to a gist a clean /tmp/klippy.log?  Make sure to issue a RESTART or FIRMWARE_RESTART command first.

markwinger, 2020-05-12:
Rebooted.  Then attacked with octoprint, then tried to open dwc2 gui.
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper/files/4612906/klippy.log)

pluuuk, 2020-05-12:
It seems your SD card path is configured for octoprint and its not pointing to where you extracted duetwebcontrol.

This is the error:
`DWC2 failed to start, no webif found in /home/pi/.octoprint/uploads/dwc2/web`

markwinger, 2020-05-12:
Thanks.  I had the path commented out and was using the one I had for octoprint.  The gui starts now.

pluuuk, 2020-05-12:
Thanks, let me know if you find issues. The fixes haven't been fully tested
yet, I hope some more users get a chance to ensure it all works properly.

manu7irl, 2020-05-12:
> Yes:
> [virtual_sdcard]
> path: ~/.octoprint/uploads/
>
> and yes
> gcode=$(sed 's/self.bytes_read = 0/self.bytes_read = 0\n self.respond_callbacks = []/g' klipper/klippy/gcode.py)
> gcode=$(echo "$gcode" | [...]

Hi my script is for scratch install. Save your printer.cfg file. Uninstall the regular klipper and reinstall it with my script klipper and dwc2 will then be installed in a structured folder called PrinterFarm. And everything will be handle from there, in Octoprint nothing is needed to change but the virtual serial endpoint which will be /tmp/printer-0 and the printer.cfg file location will be at ~/PrinterFarm/KlipperFarm/printer-0/printer-0.cfg.
If you uninstall command to remove the old init.d startup script tell me I will give you.
For follow my readme to understand more about the changes.

markwinger, 2020-05-12:
I did an overnight printer with it.  The printer came out fine but there were many notifications on the screen and I can't get rid of them.  They say:
ok B27.1/0.0 TO 25.0/0.4
Not SD Printing
ok

or
SD Not Printing
ok
ok B26:26.0/0.0 to B26:0/0.0

These messages messages all show up on the klipper console in too with a lot of NOT SD printing ok messages.  These messages seem to never stop.  The print finished 5 or six hours ago but these popup messages seem to still be printing.   There are so many the cover the right side of the screen top to bottom, covering other important info.

How do I stop them?  Can I turn off these notification popups somehow, or filter all these constant status messages.

manu7irl, 2020-05-12:
I am almost 10 hours printing without issue and no sdcard messages.

markwinger, 2020-05-12:
I also cannot run any macros from the dashboard/macros or the file management/macros.
If I specify a macro in the file management/macros I get an error:
M98 P"0:/macros/G29"
SD card not printing
ok.

I have a macro for pause and resume used for filament run out.  They work from the printer lcd control.  But if I resume using my macro I get this error:
5/12/2020, 10:02:03 AM | M98 P"0:/macros/MY_RESUME"JSON.parse: unexpected non-whitespace character after JSON data at line 1 column 22 of the JSON data

markwinger, 2020-05-12:
I also see a lot of messages: "response too long, see console"  but I see nothing in the console about them.

markwinger, 2020-05-12:
Is there a more appropriate place to report issues I see?  There are many.

pluuuk, 2020-05-12:
@markwinger, can you summarize all the issues you're seeing? I think this is the appropriate place to report issues. The ack system in gcode.py changed, so it's expected to see more messages than usual , but i haven't seen that many.

pluuuk, 2020-05-12:
About the SDcard messages, it seems it's an issue with octoprint. Can you disable octoprint and test if the messages go away? 
https://github.com/OctoPrint/OctoPrint/issues/3258

pluuuk, 2020-05-12:
I will take a look at the macro system and report back what i find.

markwinger, 2020-05-12:
How do I disable octoprint?

pluuuk, 2020-05-12:
It depends on your linux distro, but I'm guessing the commands down here should work. 

sudo systemctl stop octoprint
If you don't want it to start automatically:
sudo systemctl disable octoprint
You can re-enable it with:
sudo systemctl enable octoprint

markwinger, 2020-05-12:
That definitely stopped the endless streams of notification.  

I know the macros dont work and I'll document here other things I find odd or not working.

manu7irl, 2020-05-12:
You problably should start from scratch as I have no errors at all! I run 2
printers simultaneously. Everything is ok.

pluuuk, 2020-05-12:
@markwinger, to help with the macro issue, can you attach your macro?

markwinger, 2020-05-12:
Below is the my_pause macro.  The my_pause macro is called when out of filament and that works correctly.  I can call it from the lcd control panel but fails if I try to execute it from dwc2.  my_pause calls my_temps which I also included.

If I right click on the macro in the file management/Macro on any of the macros and click edit, I get an error:
0:/macros/MY_PAUSE

These macros are all in the printer.cfg file.  Do I to move them out of the printer.cfg file or something?

[gcode_macro my_pause]
gcode:
     ; relative XYZE
     my_temps
     G91
     M83

     ; retract filament of 25 mm up, move Z slightly upwards and
     G1 Z+5 E-10 F1000
     G1 E-50 F1000

     ; absolute XYZE
     M82
     G90

     ; move to a safe rest position, adjust as necessary
     G1 X0 Y0
     M117 Filament out
[gcode_macro my_temps]
variable_nozzle_temp: 0
gcode:
     # save nozzle temp
     SET_GCODE_VARIABLE MACRO=my_temps VARIABLE=nozzle_temp VALUE={printer.extruder.target}
     M104 S0 ; turn off hot end heater
     SET_IDLE_TIMEOUT timeout=20000

markwinger, 2020-05-12:
About pause.  If I pause using the pause button it stops and resume restarts but doe no retraction or head positioning.  The docs for it say that pause executes the pause.g system macro.  Under system I see no macros, only config.g which is printer.cfg.   I'm guessing if I want to change the pause functionality I need to create a pause macro, the current behavior is a default and there is no macro.  Is that correct?

pluuuk, 2020-05-12:
For the pause and resume behavior, the original repo had MOVE=0, which does
not move the head back to its position.

The GCODE to resume and move the head is:

RESTORE_GCODE_STATE NAME=DWC_BOTTON MOVE=1

 I think this is the default behavior for reprapfirmware. I will take a
look at your macro issue later today.

tripstar76, 2020-05-12:
Try changing the macro name to capitals......MY_PAUSE.

markwinger, 2020-05-12:
G29 is already caps and it does not work.  (long print still running so I cant change anything yet)

None of my macros can be executed or edited in the file management section.  Should I be able to edit them?

tripstar76, 2020-05-12:
@pluuuk 

Is this a typo?
line 1074:

	def cmd_M32(self, gcmd):
		self.sdcard.cmd_M23(gcmd)

Szeker, 2020-05-12:
I beleive, it is "Select file and start SD print", so first load the file => M23, than start printing => M24.

pluuuk, 2020-05-12:
@tripstar76 and @Szeker, can you provide your comments about the fixes on #75?
It's hard to keep track of the conversation.
@tripstar76, Szeker is correct. M24 starts the print, M32 is the reprapfirmware command that is natively implemented by klipper as M23. 
Regarding the macros, I can take a look and see if I broke something when re implementing the gcodes for macro management. I will provide feedback to @markwinger here.

markwinger, 2020-05-12:
Changed the macro to MY_PAUSE.  No Change.

pluuuk, 2020-05-13:
@markwinger , I have fixed the macro issue.
Please do a git pull of the latest commit on the fork in #75 .

markwinger, 2020-05-13:
Again, I don't know how git works so what commands do I issue to pull from #75.

pluuuk, 2020-05-13:
I'm assuming that you installed using my instructions, so simply running `git pull` works.

markwinger, 2020-05-13:
Progress, I can now run macros. 
I still cannot edit from file management.  Is that supported?

[16 more replies]

closed 2020-06-30 by Stephan3