Stephan3/dwc2-for-klipper #76: Before print / after print - gcode scripts?

Szeker, 2020-05-11:
Question: would it be possible to run a gcode script (defined in printer.cfg) before / after printing?

I prefer to keep things separated, therefore I like to have only object relevant g-code in the sliced gcode and machine dependent stuff stored in the printer.cfg file.
Avoiding in-place scripts in the sliced gcode also helps to keep eg. backward compatibility with existing gcodes. 

One could define how to start / finish a print in printer.cfg on a given machine - if a g-code script would be called by DWC2 before the print started / after the print finished (especially the latter).

Similar to pause/resume scripts.

manu7irl, 2020-05-11:
It is klipper related, you could make some macro like in your config.g
printer.cfg
Start_gcode
End_gcode
Each one with the gcode commands, you want to run.
Place their names in your prefered slicer.
Search on klipper gcodes about macros.
If you need an example, I can post.

Szeker, 2020-05-11:
@manu7irl  Thanks! I know I can place macros into the gcode by the slicer and execute them with Klipper, but this means I have to manually edit existing gcodes if I want to re-print them.
It's not a big deal of course, but I think it would be a more elegant and portable solution to have the option for script automation in DWC2 instead of the slicer/gcode itself.

Anyhow thanks for the comment!

Stephan3, 2020-06-30:
In my oppinion humans can do that in nearly every slicer allready. -> close

closed 2020-06-30 by Stephan3