Stephan3/dwc2-for-klipper #77: DWC2 does not start

icstra, 2020-05-15:


Avtonomys, 2020-05-15:
https://github.com/KevinOConnor/klipper/commit/787ed452c2abe3bad81ffaf732b0b2609e3534bb

icstra, 2020-05-15:
Thanks for reply. After checking, it looks like my files already get these modifications.
I’m investigating now on ports opening.

PrzemoPOL, 2020-05-15:
If you have `AttributeError: Printer instance has no attribute 'try_load_module'` in klippy.log and fresh klipper try to change line 49 in file web_dwc2.py from
`self.stepper_enable = self.printer.try_load_module(config, "stepper_enable")`
to
`self.stepper_enable = self.printer.load_object(config, "stepper_enable")`

icstra, 2020-05-15:
Thanks for reply. I had already change that line. I ´ve noticed that the symlink dwc2 to klippy was broken. I ´ve fixed that. Then I got an error message from dwc2 ( meaning that link was restored) telling me that /temp/printer was missing. So, I checked in my Raspbian and indeed there is a file named printer in tmp. Unfortunately, this file is not there in my Manjaro installation. I commented the line in dwc2 and the web server pop up in my browser. So, I progress. Of course, without serial instance, I cannot connect to my printer. I don’t understand why this file is missing. Finally, I found a walk around . I modified dwc2.py and replaced /tmp/printer with /dev/pts/0 and all seems to run. I can connect to web server and just tried a homing which well achieved. Will try later if all other functions are running well. For information, all is working with 2.1.7 dwc2. Thanks to all for your help.

closed 2020-05-15 by icstra