Stephan3/dwc2-for-klipper #78: I installed klipper + dwc2 can't start webserver

gustavya, 2020-05-15:
Hello everyone, I installed klipper+dwc2 and finally flashed a powerspec pro(same as FFCP)

I have the following error in the log file, I guess it's why I can't see the webserver.

Unhandled exception during connect
Traceback (most recent call last):
File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
self._read_config()
File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
self.load_object(config, section_config.get_name(), None)
File "/home/pi/klipper/klippy/klippy.py", line 118, in load_object
self.objects[section] = init_func(config.getsection(section))
File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1684, in load_config
return web_dwc2(config)
The RPI seems connected and the board flashed fine, since I can see movement in the lcd, like a wave of lines flowing.

Any advice?
TIA

Gus

maketo3D, 2020-05-16:
The same problem here, starting from a clean installation and with a printer.cfg working with octoprint.

Torreip, 2020-05-17:
What version of dwc are you using? readme mentions 2.1.6, is it possible you took something newer?

maketo3D, 2020-05-17:
This worked for me.
https://github.com/Stephan3/dwc2-for-klipper/pull/79

shaiss, 2020-05-20:
> This worked for me.
> #79

confirmed that works too!

closed 2020-06-30 by Stephan3