Stephan3/dwc2-for-klipper #85: web_dwc2.py 'try_load_module' error

thefabricator03, 2020-06-07:
Hi, I did a bit of searching but I got this error when trying to install dwc2-for-klipper,

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 118, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1684, in load_config
    return web_dwc2(config)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 49, in __init__
    self.stepper_enable = self.printer.try_load_module(config, "stepper_enable")
AttributeError: Printer instance has no attribute 'try_load_module'

I solved the issue by changing /klipper/klippy/extras/web_dwc2.py line:49 to
self.stepper_enable = self.printer.load_object(config, "stepper_enable")

I was able to start the interface after the change and restarting klipper.

Tomblarom, 2020-06-07:
Maybe read the other issues first or use your error "try_load_module" as search term, before creating a new one.

Currently stephan is busy and can't fix this repo. Use this fork instead: https://github.com/pluuuk/dwc2-for-klipper
It has all fixes doesn't require you to modify the klipper repo anymore.

It's equal to this PR: https://github.com/Stephan3/dwc2-for-klipper/pull/75

thefabricator03, 2020-06-07:
Ok, I used this web site as a guide on how to install dwc2 and klipper, https://klipper.info/klipper-+-dwc2-1/installing-dwc2 .

As someone new to 3d printing I was hoping it would be straight forward. I did not realize that there was a "fork" of dwc2. I will look more thoroughly next time. 

Thanks you for your help.

closed 2020-06-07 by thefabricator03