Stephan3/dwc2-for-klipper #87: Jog Command Toolbar Hidden

thefabricator03, 2020-06-12:
Hi, I am trying to get the jog command bar to appear but I cannot work out how. When I first load the DWC2 interface the box appears but then it disappears. Reading t

he manual, https://duet3d.dozuki.com/Wiki/Duet_Web_Control_Manual 
It says the interface is dynamic and features will only appear if they are configured. My issues is how do I configure the features I want to show? That part for some reason is not in the manual.
[image]
Thanks for any help.

BlackStump, 2020-06-13:
Look at your Status => OFF 
When klipper is happy you will see what you are looking for.
I suggest you have a look at the klippy.log to see what your problem is

thefabricator03, 2020-06-14:
Thanks, I have worked out that I have bigger problems. My log file is showing a MCU problem.

closed 2020-06-14 by thefabricator03