Stephan3/dwc2-for-klipper #94: Cannot print gcode from sub directories.

changedsoul, 2020-07-04:
[klipper.log](https://github.com/Stephan3/dwc2-for-klipper/files/4873806/klipper.log)
When I try and print a file in a sub directory I get an error saying it cannot open the file.
DWC2:75281.96 - 10.0.0.44 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FCalibration%2Fsquare_tower.gcode%22 virtual_sdcard file open Traceback (most recent call last): File "/home/pi/PrinterFarm/KlipperFarm/klipper/klippy/extras/virtual_sdcard.py", line 102, in cmd_M23 try: KeyError: 'gcodes/calibration/square_tower.gcode' failed: M32Unable to open file

I can print it from the root of the gcodes folder, so I dont know why its failing.
Does klipper not support printing from sub folders? Perhaps DWC2 is sending some garbage over to klipper? I dont really know how the two interact.

robthide37, 2020-07-10:
try remaming with no spaces or special characters. I have had issue where i got the same error when trying to send from superslicer and getting rid of the characters fixed it.

changedsoul, 2020-07-10:
I tried that, I noticed in the log it looked like it was changing everything to lower case. I still get an error with everything lower case and no spaces.

Stephan3, 2020-07-20:
This works:
INFO:root:DWC2:49032.11 - 192.168.10.240 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2Ftest_dir%2Freprint.gcode%22
So printing from sdcard/gcodes/test_dir/reprint.gcode is working as ecxpected

So this does too.
INFO:root:DWC2:51000.85 - 192.168.10.240 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2Fcalibration%2Fsquare_tower.gcode%22
sdcard/gcodes/calibration/square_tower.gcode

can this be closed?

changedsoul, 2020-07-20:
So yours works fine without changing anything in the code? Very strange. Something I should have checked before but did not was to see what the permissions were on the sub directory. I can create and delete files in it, so I can only assume permissions are ok. 
Other than that, I do not know what else to look for as to why my sub directory printing does not work.

Stephan3, 2020-07-20:
yes. thats a good point! It has to be the user that runs klipper.

changedsoul, 2020-07-21:
Well scratch that on permissions. They are set correctly....or at least they are set the same as the main directories permissions. So now what else could it be? I tried setting the sub directory and files in it with 777 perms and it still fails with "Cannot open file".

Stephan3, 2020-07-21:
try

chown -r pi:pi ~sdcard

if you are on a pi. replace with user and group of you klipper user

changedsoul, 2020-07-22:
The sub directory is already owned by pi:pi and its got full permissions and it still does not work.
Any other ideas? Is there something I can enable to get more detailed debug info as to why its not working?

Stephan3, 2020-07-22:
Thats the function in charge for it. I really tried hard the last two hours to provoke an erreor. I was not able to.

https://github.com/Stephan3/dwc2-for-klipper/blob/ad8e08088685535d6be82b8aa18c992f46d85643/web_dwc2.py#L1060

You might try to put some logging.info(somvartocheck) here and there and try to find out.

changedsoul, 2020-07-24:
> Thats the function in charge for it. I really tried hard the last two hours to provoke an erreor. I was not able to. [...]

Ok great, I’ll give that a try. I was looking for a way to output debug info and could not figure out how its outpouring info. I’ll use the logging.info and try it out.

Stephan3, 2020-07-24:
Klipper logfile....

changedsoul, 2020-07-24:
> Klipper logfile....

Sorry I wasted your time. I was changing the wrong file. I thought the file in dwc2-for -klipper was symlinked to the klippy/extras folder.....it was not. I was changing the wrong file. Which makes me wonder if the script for updating dwc2 is even updating the correct file. 
I am going to remove it all and install directly from github and see if it solves the problem. I was using the install script someone wrote to make updating klipper and dwc easier. Perhaps this is whats causing the issues. 
Ill keep you posted. If it solves the issue, Ill report back and close this issue.

changedsoul, 2020-07-24:
So I updated to the latest version of Klipper and dwc2 without using the helper install script and printing from sub directories is working.
I appreciate you taking the time to help try and figuring this out, even though I wasn't able to find that actual fault other than just reinstalling.

closed 2020-07-24 by changedsoul