Stephan3/dwc2-for-klipper #96: fan speed

robthide37, 2020-07-10:
fan speed stopped showing with the newest commit. i can still control the fan so no big deal but it won't show what it actually is. My only fear is if i do change the fan speed i can't tell how persistent it is so won't know if the slicer changed it again or not unless its almost full speed since i am running a very quiet 5015

Stephan3, 2020-07-15:
i see. But cant explain. the percent value got handed over as ecxpected.

exxperto, 2020-07-15:
It's happened after DuetWebControl 2.0.7
[displaying the fan speed in the web interface](https://github.com/Duet3D/DuetWebControl/issues/241#issuecomment-624615928)

robthide37, 2020-07-16:
same for me downgrade works . Also superslicer regex worked when i switched prusaslicer to superslicer in regex but now for some reason layer count and sometimes height is way off i wonder if that is also only 3.1 i have to downgrade again and check ill post issue when i do verify more specifics.

robthide37, 2020-07-16:
also on my rpi screen fan speed works its only across actual network it doesn't . i tried chrome and edge to rule that out

robthide37, 2020-07-19:
I think i will close because it seems to be issue with dwc not this py code

closed 2020-07-19 by robthide37