Stephan3/dwc2-for-klipper #97: Gcode upload inserted random characters into gcode file

ju-sti, 2020-07-11:
The printing of a job stopped suddenly today with this error:[image]

I uploaded the printed gcode file with dwc to my printer.
After I downloaded the gcode file from the printer to check what is wrong I noticed the issue immediately:
[image]

The original file as created by Cura has proper content.  Hostname of the RPi where dwc2 for klipper is running is "raspi-print".

So I very much suspect that either uploading or printing the file in dwc has corrupted the file by inserting weird bytes and the hostname for some reason.

ju-sti, 2020-07-11:
* I'm using dwc2-for-klipper version with the "add mutex to gcode ecxecution. try to fix the one command behind" change
* Uploading the gcode and downloading it again did not lead to the same error. The file did not have changed content as far as I can see.

Stephan3, 2020-07-15:
okay thats odd. i was not able to reproduce this.

Stephan3, 2020-07-20:
I tried today to provoke this using a pi zero and i had just thrown things at it as much as possible. No matter how hard i try, i am not able to make this happen.
Did that ever happen again?

ju-sti, 2020-07-20:
Thanks for trying. I "uploaded and started" at least 30 times since that error occurred but the error did not happen again.

Stephan3, 2020-07-22:
I definitley cant reproduce it. Sorry.

closed 2020-07-22 by Stephan3