Stephan3/dwc2-for-klipper #98: DWC UI working, won't connect to printer?

rlesac, 2020-07-15:
Hi!
Installed 3.1.1 DWC, configured the printer.cfg, I can get the UI to show up in the browser, but the printer is always off.
I see no issues in the klippy.log
Octoprint works fine, If I connect to Octoprint, I can see the console data in DWC, but that's about it.

The only thing that seems to work is loading the height map.

The status panel is "twitchy", it will change height every few seconds.

Any suggestions?

Stephan3, 2020-07-15:
off is detected if printer state is not ready. have you tried a firmware_restart ?

rlesac, 2020-07-15:
> off is detected if printer state is not ready. have you tried a firmware_restart ?

I tried, the commands do not go through. There is nothing in the klippy.log
If I try Octoprint, Octoprint connects just fine.
Here's my klippy.log with the printer.cfg

[dwc-debug1.log](https://github.com/Stephan3/dwc2-for-klipper/files/4925555/dwc-debug1.log)

rlesac, 2020-07-19:
Any ideas what to try next?

Stephan3, 2020-07-20:
I am verry sorry. I just picked your config and put it on my testmachine. I had to comment out display and bltouch ad its not there on the testsystem. It just worked. Can you try to comment out display and bltoch and report back?

saxmfone1, 2020-08-24:
Ah, thank you, that solves it.