Stephan3/dwc2-for-klipper #99: regex support superslicer

robthide37, 2020-07-19:
for some reason object height is showing as the first layer height . I had switched out superslicer and slicer pe edition in past and it worked but when updated it overwrote my edits and now for some reason that its added it isn't reading the layer height .

Stephan3, 2020-07-20:
please provide the gcode file. Thanks

Stephan3, 2020-07-22:
meanwhile i downloaded superslicer and checked things out. regex is working as ecxpected. Thanks. 

-> close

closed 2020-07-22 by Stephan3

robthide37, 2020-07-26:
[yoda.curved.gcode.txt](https://github.com/Stephan3/dwc2-for-klipper/files/4978003/yoda.curved.gcode.txt)
here is one

robthide37, 2020-07-26:
i missed you response but if you need more i can make more it seems random. fist i thought it was if the first layer was dif but that doesn't seem to be the case. Also i made sure that no adaptive layers or anything were on .

robthide37, 2020-07-26:
[image]
[MeshBody27.gcode.txt](https://github.com/Stephan3/dwc2-for-klipper/files/4978832/MeshBody27.gcode.txt)
another example of it reading wrong