Stephan3/dwc2-for-klipper #100: Not compatible with Klipper latests commits.

jairbj, 2020-08-07:
Hi friend, after spending this afternoon setting up for my first time a pure Raspian Lite + Klipper + DWC (on Zero W) I found that the script was not working.

According I could see this was due to Klipper latest updates on G-Code parser.

After testing I finally got it working when I checked out the commit b9f48e6 (which was submitted 2 days ago).

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 186, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 251, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 282, in _dispatch_loop
    self._fds[fd](eventtime)
  File "/home/pi/klipper/klippy/gcode.py", line 722, in _process_data
    self.gcode._process_commands(pending_commands)
  File "/home/pi/klipper/klippy/gcode.py", line 281, in _process_commands
    gcmd.ack()
  File "/home/pi/klipper/klippy/gcode.py", line 31, in ack
    self.respond_raw(ok_msg)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1575, in respond_raw
    os.write(self.gcode.fd, msg+"\n")
AttributeError: GCodeParser instance has no attribute 'fd'
```

Stephan3, 2020-08-07:
looks like this commit
https://github.com/KevinOConnor/klipper/commit/0d930bf8718c316ec0f40440d102ffd4fa66b91e#diff-41fbedb77d3bf5215035bbe0c12a90a9

jairbj, 2020-08-07:
Hi @Stephan3, this shouldn't be the only one.

When I commented these 2 lines on web_dwc2.py it started Klipper but DWC showed other errors related to GCODE parsing (which I don't have the logs right now).
```
#	prevent serial overflow if octoprint disconnects
			if self.serial.in_waiting < 500:
				os.write(self.gcode.fd, msg+"\n")
```

Stephan3, 2020-08-07:
checkout corrent master

closed 2020-08-07 by Stephan3