Stephan3/dwc2-for-klipper #101: Connection status switches every second, AttributeError

Zapfenechse, 2020-08-24:
I just updated klipper to the newest version with the KIAUH script from Github published by th33xitus.
After it I can't use the DWC 2 web interface anymore.
It only says: connection esteblished and directly after it: conncection interrupted
A tried multiple fresh installation of klipper and DWC but the problem is still there.

In the klippy.log i found the following Error:
`AttributeError: GCodeDispatch instance has no attribute 'get_status'`

But I have no clue how to fix it.
My klippy.log is attached to this issue.

My used hardware:
Ender 3 Pro
Raspberry Pi4
SKR 1.4 Board with TMC 2209

Version:
klipper: 67632105 
DWC2 for klipper: f783180e
DWC2 web UI: 3.1.1

[klippy.log](https://github.com/KevinOConnor/klipper/files/5114832/klippy.log)

Stephan3, 2020-08-24:
latest master is broken. for now rollback few days:

```
cd klipper
git reset --hard 560d48d
sudo service klipper restart
```

Zapfenechse, 2020-08-24:
Thank you very much!

The roll back fixed it for now.

CappyT, 2020-08-31:
Still experiencing this error? Can you indicate where the issue is? I can try to make a pull request to fix it.

apphead, 2020-09-06:
While Stephans suggestion https://github.com/Stephan3/dwc2-for-klipper/issues/101#issuecomment-679213417 does the trick, it is not a permanent solution. Is it something that needs fixing on klipper or DWC side of things? Or both?

Stephan3, 2020-09-06:
i wont fix this anymore. meanwhile i have a upcomming rewrite which runs offthread and uses klippers new unixsocket for communication. This will be much easier to maintain.

apphead, 2020-09-06:
That's great news, Stephan. I would let people know in your readme that DWC currently does not work out of the box and that there is update in progress or if they wanted it so badly, then they need to run [the reset above](https://github.com/Stephan3/dwc2-for-klipper/issues/101#issuecomment-679213417).

ajalberd, 2020-09-11:
Is dwc2-for-klipper-socket working currently? Should we switch to it?

Stephan3, 2020-09-11:
yes, site is updated
https://github.com/Stephan3/dwc2-for-klipper-socket

closed 2020-09-11 by Stephan3