Stephan3/dwc2-for-klipper #104: Works with non-Duet hardware?

sixtyfive, 2020-08-30:
Could it be mentioned explicitly somewhere whether this makes DWC work with non-Duet hardware? In my case, I'd love to use it with my Klipper-enabled Ender 2 because it seems to be much nicer than OctoPrint in a number of ways, but I'm hesitant to go through the trouble of figuring out DWC because it says it's for Duet hardware only. Does dwc2-for-klipper change that?

trevjonez, 2020-08-30:
it can work but isn't quite as fluid of an experience as what you should expect with duet hardware. This project is basically a web server shim between klipper and the dwc2 code. Due to it being an adaptation that is not directly supported by klipper or the dwc project, it can be more work to keep up to date and functioning.

All that said I have used this project with three klipper machines for over a year to great effect and only recently stopped using it in favor of just using a custom terminal experience built on top of https://github.com/tmuxinator/tmuxinator . It is nowhere near as nice as dwc in terms of features, but I have grown tired of trying to keep the three machines all in sync with one another while making various hardware and software changes unrelated to dwc2-for-klipper.

another alternative that is up and coming is https://github.com/meteyou/mainsail along with https://github.com/Arksine/moonraker to support it.

moonraker has been getting great support from within klipper and I expect this combo will eventually be a far better choice due to the more direct support within klipper. Though I believe it should be possible to build a new version of this project on top of moonraker to enable a good and probably more decoupled maintenance experience with dwc2. Likely true for any octoprint integration as well.

disclaimer: I am just a bystander dumping my opinion into the internet

sixtyfive, 2020-08-30:
Much appreciated, thank you! :)

closed 2020-08-30 by sixtyfive